nvmd/nixos-raspberrypi #1: Binary cache issue

ametis70, 2024-11-10:
Hi, I'm fairly new to NixOS and I'm trying to use a flake with the following inputs:

```nix
{
  inputs = {
    nixpkgs.url = "github:nixos/nixpkgs/release-24.05";
    nixpkgs-unstable.url = "github:nixos/nixpkgs/nixos-unstable";

    nixos-raspberrypi = {
      url = "github:nvmd/nixos-raspberrypi";
      inputs = {
        nixpkgs.follows = "nixpkgs";
        nixpkgs-unstable.follows = "nixpkgs-unstable";
      };
    };
  };
  nixConfig = {
    extra-substituters = [ "https://nixos-raspberrypi.cachix.org" ];
    extra-trusted-public-keys = [
      "nixos-raspberrypi.cachix.org-1:4iMO9LXa8BqhU+Rpg6LQKiGa2lsNh/j2oiYLNOQ5sPI="
    ];
  };
}
```

But when I try to run `nix flake lock`, I'm getting the following warning:
```
warning: input 'nixos-raspberrypi' has an override for a non-existent input 'nixpkgs-unstable'
```

I don't know if it is realted to my problem, but whe I try to deploy nixos-rapsberry for the first time to a remote RPi 4 2GB using:

```sh
nixos-rebuild switch --flake . --verbose --target-host <user>@<host> --build-host <user>@<host> --use-remote-sudo 
```

But I cannot make it fetch Linux and instead it tries to compile it but eventually runs out of RAM. This RPi is already running NixOS 24.04 but I installed it following the guide in [nixos.dev](https://wiki.nixos.org/wiki/NixOS_on_ARM/Raspberry_Pi_4).

nvmd, 2024-11-10:
Hi, Ian.

Thank you for bringing up this issue!
Unfortunately, the README file was slightly outdated – the flake now doesn't have `nixpkgs-unstable` input anymore (it currently tracks `nixkpgs-unstable` as its `nixkpgs`, though). Please see `README` for more details with an updated example.

As for the binary cache, to debug the problem,
- Update the inputs with `nix flake lock --update-input nixos-raspberrypi` or equivalent, as I've just bumped it to the latest `nixpkgs-unstable` and [pushed the build results](https://github.com/nvmd/nixos-raspberrypi/blob/main/devshells/nix-build-to-cachix.nix#L41) to the cache
- Try setting `nixos-raspberrypi.inputs.nixpkgs.follows = "nixpkgs-unstable";`
- You could try not to override `nixos-raspberrypi` inputs (i.e. remove `nixos-raspberrypi.inputs.*.follows` lines altogether), maybe the version in the cache and the version in your own flake is too different. Will it work?
- Is your user `trusted`? Only trusted users can use flake substituters. Alternatively, you could add this cache to system-wide substituters, as adding the user to `trusted` implies certain security risks.

ametis70, 2024-11-11:
Ok, I think my problem was related to the the `lib.nixosSystem` call, because I was using `nixpkgs.lib.nixosSystem` instead of `nixpkgs-unstable.lib.nixosSystem`. I also updated the my nixpkgs-unstable to the same ref that used in this flake. Now, I don't have to build Linux, so I think it's using the version from cachix, but is still failing if I try to do a remote build:

```
nixos-rebuild boot --flake . --verbose --target-host trovadores@192.168.10.32 --use-remote-sudo --build-host localhost -j 1
$ ssh trovadores@192.168.10.32 cat /proc/sys/kernel/hostname
$ nix --extra-experimental-features nix-command flakes build --out-link /tmp/nixos-rebuild.BuGbIc/nixos-rebuild .#nixosConfigurations."trovadores-rpi4".config.system.build.nixos-rebuild --verbose -j 1
warning: Git tree '/home/ametis70/Sandbox/nix' is dirty
Using saved setting for 'extra-substituters = https://nixos-raspberrypi.cachix.org' from ~/.local/share/nix/trusted-settings.json.
Using saved setting for 'extra-trusted-public-keys = nixos-raspberrypi.cachix.org-1:4iMO9LXa8BqhU+Rpg6LQKiGa2lsNh/j2oiYLNOQ5sPI=' from ~/.local/share/nix/trusted-settings.json.
$ exec /nix/store/573nq0gi18am9s4sqdqazrhnkvqhvn8g-nixos-rebuild/bin/nixos-rebuild boot --flake . --verbose --target-host trovadores@192.168.10.32 --use-remote-sudo --build-host localhost -j 1
$ ssh trovadores@192.168.10.32 cat /proc/sys/kernel/hostname
building the system configuration...
Building in flake mode.
$ nix --extra-experimental-features nix-command flakes eval --raw .#nixosConfigurations."trovadores-rpi4".config.system.build.toplevel.drvPath --verbose -j 1
warning: Git tree '/home/ametis70/Sandbox/nix' is dirty
Using saved setting for 'extra-substituters = https://nixos-raspberrypi.cachix.org' from ~/.local/share/nix/trusted-settings.json.
Using saved setting for 'extra-trusted-public-keys = nixos-raspberrypi.cachix.org-1:4iMO9LXa8BqhU+Rpg6LQKiGa2lsNh/j2oiYLNOQ5sPI=' from ~/.local/share/nix/trusted-settings.json.
Running nix with these NIX_SSHOPTS:  -o ControlMaster=auto -o ControlPath=/tmp/nixos-rebuild.s3eZPt/ssh-%n -o ControlPersist=60
$ nix --extra-experimental-features nix-command flakes copy --verbose -j 1 --derivation --to ssh://localhost /nix/store/9mhqsrlkv734n612hic8493fzmq7wcdg-nixos-system-trovadores-rpi4-24.11.20241105.4aa3656.drv
warning: The interpretation of store paths arguments ending in `.drv` recently changed. If this command is now failing try again with '/nix/store/9mhqsrlkv734n612hic8493fzmq7wcdg-nixos-system-trovadores-rpi4-24.11.20241105.4aa3656.drv^*'
copying 0 paths...
$ ssh -o ControlMaster=auto -o ControlPath=/tmp/nixos-rebuild.s3eZPt/ssh-%n -o ControlPersist=60 localhost nix-store -r /nix/store/9mhqsrlkv734n612hic8493fzmq7wcdg-nixos-system-trovadores-rpi4-24.11.20241105.4aa3656.drv --verbose -j 1
querying info about '/nix/store/ksyhna6mr31k3skq9q82zc7ghwwldrlk-device-tree-overlays' on 'https://cache.nixos.org'...
querying info about '/nix/store/yah6vq1zr8cwihpkh21f6lfa5yhqya09-initrd-linux_rpi-bcm2711-6.6.51-stable_20241008' on 'https://cache.nixos.org'...
downloading 'https://cache.nixos.org/ksyhna6mr31k3skq9q82zc7ghwwldrlk.narinfo'...
downloading 'https://cache.nixos.org/yah6vq1zr8cwihpkh21f6lfa5yhqya09.narinfo'...
querying info about '/nix/store/yah6vq1zr8cwihpkh21f6lfa5yhqya09-initrd-linux_rpi-bcm2711-6.6.51-stable_20241008' on 'https://nix-community.cachix.org'...
downloading 'https://nix-community.cachix.org/yah6vq1zr8cwihpkh21f6lfa5yhqya09.narinfo'...
querying info about '/nix/store/ksyhna6mr31k3skq9q82zc7ghwwldrlk-device-tree-overlays' on 'https://nix-community.cachix.org'...
downloading 'https://nix-community.cachix.org/ksyhna6mr31k3skq9q82zc7ghwwldrlk.narinfo'...
these 4 derivations will be built:
  /nix/store/drvqq6bsp38w8rvkb41ghcm8qh76w8qr-device-tree-overlays.drv
  /nix/store/s0sajikpjv6000bv0mdysl8z8m5xhs5y-initrd-linux_rpi-bcm2711-6.6.51-stable_20241008.drv
  /nix/store/w40pmddsyf4a4qk71hrj81mm7zmqi62c-boot.json.drv
  /nix/store/9mhqsrlkv734n612hic8493fzmq7wcdg-nixos-system-trovadores-rpi4-24.11.20241105.4aa3656.drv
building '/nix/store/drvqq6bsp38w8rvkb41ghcm8qh76w8qr-device-tree-overlays.drv'...
Processing source device tree overlays/hat_map.dtb...
Traceback (most recent call last):
  File "/nix/store/206l37y6cvcwl3ck1ppwgnxn4cnm818i-apply_overlays.py", line 102, in <module>
    main()
  File "/nix/store/206l37y6cvcwl3ck1ppwgnxn4cnm818i-apply_overlays.py", line 80, in main
    dt_compatible = get_compatible(dt)
                    ^^^^^^^^^^^^^^^^^^
  File "/nix/store/206l37y6cvcwl3ck1ppwgnxn4cnm818i-apply_overlays.py", line 28, in get_compatible
    return set(fdt.getprop(root_offset, "compatible").as_stringlist())
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/nix/store/g70z9kpkh2xf0h8zxk9ycvr8ydm9z71r-python3-3.12.7-env/lib/python3.12/site-packages/libfdt.py", line 450, in getprop
    pdata = check_err_null(fdt_getprop(self._fdt, nodeoffset, prop_name),
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/nix/store/g70z9kpkh2xf0h8zxk9ycvr8ydm9z71r-python3-3.12.7-env/lib/python3.12/site-packages/libfdt.py", line 153, in check_err_null
    raise FdtException(val)
libfdt.FdtException: pylibfdt error -1: FDT_ERR_NOTFOUND
error: builder for '/nix/store/drvqq6bsp38w8rvkb41ghcm8qh76w8qr-device-tree-overlays.drv' failed with exit code 1;
       last 10 log lines:
       >                     ^^^^^^^^^^^^^^^^^^
[... cut ...]
       >     raise FdtException(val)
       > libfdt.FdtException: pylibfdt error -1: FDT_ERR_NOTFOUND
       For full logs, run 'nix log /nix/store/drvqq6bsp38w8rvkb41ghcm8qh76w8qr-device-tree-overlays.drv'.
error: 1 dependencies of derivation '/nix/store/9mhqsrlkv734n612hic8493fzmq7wcdg-nixos-system-trovadores-rpi4-24.11.20241105.4aa3656.drv' failed to build
```

Maybe this is because I'm trying to do the build locally (using binfmt in a x86_64 host)? I'll try buliding in the RPi next and I'll let you know if it works

nvmd, 2024-11-11:
I figure, you're either using `argonone` module or some other device tree overlay.

This is caused by the recent change in `nixpkgs` in `deviceTree.applyOverlays` (NixOS/nixpkgs#341883), which seem to have a regression, causing it to break on raspberry-specific `overlay_map.dtb` and `hat_map.dtb`.
- You could either disable `argonone` module and other overlays
- or, if you're feeling adventurous, you could try using https://github.com/nvmd/nixpkgs/tree/apply-overlays-no-compatible (it's rebased on top of the current `nixpkgs-unstable`), it _seems_ to work, but I didn't yet have a possibility to properly test it. It might break the device tree, making your system non-bootable, so have a rescue plan if that happens.

ametis70, 2024-11-11:
I was indeed trying to use  the `raspberry-pi-4.case-argonone`. Without it, the build succeeded but until it got to the point where the firmware should be copied:

```
nixos-rebuild boot --flake . --verbose --target-host trovadores@192.168.10.32 --use-remote-sudo --build-host localhost -j 1
$ ssh trovadores@192.168.10.32 cat /proc/sys/kernel/hostname
$ nix --extra-experimental-features nix-command flakes build --out-link /tmp/nixos-rebuild.KvHPo7/nixos-rebuild .#nixosConfigurations."trovadores-rpi4".config.system.build.nixos-rebuild --verbose -j 1
warning: Git tree '/home/ametis70/Sandbox/nix' is dirty
Using saved setting for 'extra-substituters = https://nixos-raspberrypi.cachix.org' from ~/.local/share/nix/trusted-settings.json.
Using saved setting for 'extra-trusted-public-keys = nixos-raspberrypi.cachix.org-1:4iMO9LXa8BqhU+Rpg6LQKiGa2lsNh/j2oiYLNOQ5sPI=' from ~/.local/share/nix/trusted-settings.json.
$ exec /nix/store/573nq0gi18am9s4sqdqazrhnkvqhvn8g-nixos-rebuild/bin/nixos-rebuild boot --flake . --verbose --target-host trovadores@192.168.10.32 --use-remote-sudo --build-host localhost -j 1
$ ssh trovadores@192.168.10.32 cat /proc/sys/kernel/hostname
building the system configuration...
Building in flake mode.
$ nix --extra-experimental-features nix-command flakes eval --raw .#nixosConfigurations."trovadores-rpi4".config.system.build.toplevel.drvPath --verbose -j 1
warning: Git tree '/home/ametis70/Sandbox/nix' is dirty
Using saved setting for 'extra-substituters = https://nixos-raspberrypi.cachix.org' from ~/.local/share/nix/trusted-settings.json.
Using saved setting for 'extra-trusted-public-keys = nixos-raspberrypi.cachix.org-1:4iMO9LXa8BqhU+Rpg6LQKiGa2lsNh/j2oiYLNOQ5sPI=' from ~/.local/share/nix/trusted-settings.json.
Running nix with these NIX_SSHOPTS:  -o ControlMaster=auto -o ControlPath=/tmp/nixos-rebuild.BvzUVQ/ssh-%n -o ControlPersist=60
$ nix --extra-experimental-features nix-command flakes copy --verbose -j 1 --derivation --to ssh://localhost /nix/store/0rv0p54wnh61qhc1dmbwz2zxfqxwpqc7-nixos-system-trovadores-rpi4-24.11.20241105.4aa3656.drv
warning: The interpretation of store paths arguments ending in `.drv` recently changed. If this command is now failing try again with '/nix/store/0rv0p54wnh61qhc1dmbwz2zxfqxwpqc7-nixos-system-trovadores-rpi4-24.11.20241105.4aa3656.drv^*'
copying 0 paths...
$ ssh -o ControlMaster=auto -o ControlPath=/tmp/nixos-rebuild.BvzUVQ/ssh-%n -o ControlPersist=60 localhost nix-store -r /nix/store/0rv0p54wnh61qhc1dmbwz2zxfqxwpqc7-nixos-system-trovadores-rpi4-24.11.20241105.4aa3656.drv --verbose -j 1
warning: you did not specify '--add-root'; the result might be removed by the garbage collector
$ ssh -o ControlMaster=auto -o ControlPath=/tmp/nixos-rebuild.BvzUVQ/ssh-%n -o ControlPersist=60 localhost nix-copy-closure --verbose -j 1 --to trovadores@192.168.10.32 /nix/store/37hz5gqqswl77y86rhn91zfrl2w4h10f-nixos-system-trovadores-rpi4-24.11.20241105.4aa3656
copying 0 paths...
$ ssh -o ControlMaster=auto -o ControlPath=/tmp/nixos-rebuild.BvzUVQ/ssh-%n -o ControlPersist=60 -t trovadores@192.168.10.32 sudo nix-env -p /nix/var/nix/profiles/system --set /nix/store/37hz5gqqswl77y86rhn91zfrl2w4h10f-nixos-system-trovadores-rpi4-24.11.20241105.4aa3656
[sudo] password for trovadores: 
Shared connection to 192.168.10.32 closed.
$ ssh -o ControlMaster=auto -o ControlPath=/tmp/nixos-rebuild.BvzUVQ/ssh-%n -o ControlPersist=60 -t trovadores@192.168.10.32 sudo systemd-run -E LOCALE_ARCHIVE -E NIXOS_INSTALL_BOOTLOADER= --collect --no-ask-password --pipe --quiet --service-type=exec --unit=nixos-rebuild-switch-to-configuration --wait true
[sudo] password for trovadores: 
Shared connection to 192.168.10.32 closed.
Using systemd-run to switch configuration.
$ ssh -o ControlMaster=auto -o ControlPath=/tmp/nixos-rebuild.BvzUVQ/ssh-%n -o ControlPersist=60 -t trovadores@192.168.10.32 sudo systemd-run -E LOCALE_ARCHIVE -E NIXOS_INSTALL_BOOTLOADER= --collect --no-ask-password --pipe --quiet --service-type=exec --unit=nixos-rebuild-switch-to-configuration --wait /nix/store/37hz5gqqswl77y86rhn91zfrl2w4h10f-nixos-system-trovadores-rpi4-24.11.20241105.4aa3656/bin/switch-to-configuration boot
[sudo] password for trovadores: 
uboot-builder: -f /boot/firmware -b /boot -c /nix/store/37hz5gqqswl77y86rhn91zfrl2w4h10f-nixos-system-trovadores-rpi4-24.11.20241105.4aa3656
copying firmware...
using vendor firmware from /nix/store/6n0lncri0b46rys5a86bvafvahczkizp-raspberrypi-firmware-1.20241008/share/raspberrypi/boot
cp: cannot create regular file '/boot/firmware/bcm2708-rpi-b.dtb.tmp': No such file or directory
Failed to install bootloader
Shared connection to 192.168.10.32 closed.
warning: error(s) occurred while switching to the new configuration
```

Maybe I should change the current bootloader or mount the firmware partition before attempting to deploy `nixos-raspberry`?

This is the `configuration.nix` for the current deployed system:

```nix
{ config, pkgs, lib, ... }:
 
{
  nix.settings.experimental-features = [
    "nix-command"
    "flakes"
  ];
 
  boot = {
    kernelPackages = pkgs.linuxKernel.packages.linux_rpi4;
    initrd.availableKernelModules = [ "xhci_pci" "usbhid" "usb_storage" ];
    loader = {
      grub.enable = false;
      generic-extlinux-compatible.enable = true;
    };
  };
 
  fileSystems = {
    "/" = {
      device = "/dev/disk/by-label/NIXOS_SD";
      fsType = "ext4";
      options = [ "noatime" ];
[... 61 lines cut ...]
    users = {
      root.initialPassword = "root";
      "${specialArgs.host.username}" = {
        isNormalUser = true;
        extraGroups = [
          "wheel"
          "networkmanager"
        ];
      };
    };
  };

  fileSystems = {
    "/" = {
      device = "/dev/disk/by-label/NIXOS_SD";
      fsType = "ext4";
      options = [ "noatime" ];
    };
  };

  system.stateVersion = "24.11";
}
```

This time I'm only using the `raspberry-pi-4.base` and `raspberry-pi-4.display-vc4` NixOS modules.

nvmd, 2024-11-12:
> Maybe I should change the current bootloader or mount the firmware partition before attempting to deploy `nixos-raspberry`?

Yes, `FIRMWARE` partition on the target system must be mounted to `/boot/firmware`

ametis70, 2024-11-12:
Great! Adding the folllowing worked:
```nix
fileSystems."/boot/firmware" = {
  device = "/dev/disk/by-label/FIRMWARE";
  fsType = "vfat";
};
```
I was able to build and switch to the new configuration.

I think we can mark this as resolved. I will now try installing a few things a see if they work correctly with the full KMS driver, like Kodi, Retroarch and Moonlight QT and I let you know if I encounter any issue.

closed 2024-11-12 by nvmd

francocalvo, 2025-08-27:
Hey, @nvmd, wanted to say THANKS for the work done here.
If only, I'd say it would be cool to add the debugging steps for missing the cache to the README.

I was missing it for the kernel which was taking too much even in my main rig, and looking on how to solve it, I found this. In my case it was the usage of `nix flake lock --update-input nixos-raspberrypi`. 

Thanks a lot!