nvmd/nixos-raspberrypi #2: Quick question

devinlittle, 2024-12-21:
So instead of using nix-community/raspberry-pi-nix I could use this?

nvmd, 2024-12-22:
Hi @devinlittle. Yes, you absolutely could

devinlittle, 2024-12-22:
Hi, @nvmd,
Thank you for your response and for clarifying! I’m still new to NixOS and flakes, so I apologize if this is a basic question. Would it be possible to provide an example of how to use Disko for setting up the partitions and configuring the boot process in this flake?

I really appreciate your time and help as I learn. Let me know if there are any specific resources I should check out!

Thank you!

nvmd, 2024-12-23:
Absolutely no problem, you're welcome. Yes, this repo currently lacks an example of the minimal but complete system configuration. Good news is that I was planning to finish one in the coming days. I'll let you know.

There's no one-stop source of information, unfortunately, you'll likely need to look into a multitude to compose your solution. At least I don't remember one from the top of my head.
NixOS wiki is one of the best, though it's page on ARM and RaspberryPi is, for my taste, too complicated right now, describing solutions with too much manual labor, etc.

devinlittle, 2025-01-27:
I've re opened this because I've done a bunch of research on nixos however I still can't seem to understand how to use this and I'm just wondering if you could upload a repo or anything that uses this so I can try to adapt and use it

nvmd, 2025-01-30:
Hi, I think I'll be able to add some minimal configuration examples to this repo in the coming days.

SolidRhino, 2025-02-11:
@nvmd Can you please add an example for this flake?

nvmd, 2025-04-20:
Demo configurations deployable with `nixos-anywhere` are now available from https://github.com/nvmd/nixos-raspberrypi-demo.
Configurations for sd-images, suitable to be used as an "installer", are available in `nixosConfigurations` of this flake.

closed 2025-04-20 by nvmd