nvmd/nixos-raspberrypi #3: Issues building the SD Image

mrsteakhouse, 2025-02-27:
Good day, thank you for your work in the subject.

I'm currently trying to implement the module into my configuration. I have multiple hosts configured through the snowfall lib.

When cross compiling using the sd image target `nix build -L .#nixosConfigurations.host.config.system.build.sdImage --system aarch64-linux` i get an error regarding the removed options:

```
error: The option `boot.loader.raspberryPi' in module `/nix/store/zbvf6xwri9kvf42xl3vai3mx8jry6ax8-source/nixos/modules/rename.nix' would be a parent of the following options, but its type `<no description>' does not support nested options.
 - option(s) with prefix `boot.loader.raspberryPi.bootPath' in module `/nix/store/szyjiswb0gfzfss63m1p3a29zqf8kxzn-source/modules/system/boot/loader/raspberrypi'
 - option(s) with prefix `boot.loader.raspberryPi.bootPopulateCmd' in module `/nix/store/szyjiswb0gfzfss63m1p3a29zqf8kxzn-source/modules/system/boot/loader/raspberrypi'
 - option(s) with prefix `boot.loader.raspberryPi.bootloader' in module `/nix/store/szyjiswb0gfzfss63m1p3a29zqf8kxzn-source/modules/system/boot/loader/raspberrypi'
 - option(s) with prefix `boot.loader.raspberryPi.enable' in module `/nix/store/szyjiswb0gfzfss63m1p3a29zqf8kxzn-source/modules/system/boot/loader/raspberrypi'
 - option(s) with prefix `boot.loader.raspberryPi.firmwarePackage' in module `/nix/store/szyjiswb0gfzfss63m1p3a29zqf8kxzn-source/modules/system/boot/loader/raspberrypi'
 - option(s) with prefix `boot.loader.raspberryPi.firmwarePath' in module `/nix/store/szyjiswb0gfzfss63m1p3a29zqf8kxzn-source/modules/system/boot/loader/raspberrypi'
 - option(s) with prefix `boot.loader.raspberryPi.firmwarePopulateCmd' in module `/nix/store/szyjiswb0gfzfss63m1p3a29zqf8kxzn-source/modules/system/boot/loader/raspberrypi'
 - option(s) with prefix `boot.loader.raspberryPi.ubootPackage' in module `/nix/store/szyjiswb0gfzfss63m1p3a29zqf8kxzn-source/modules/system/boot/loader/raspberrypi'
 - option(s) with prefix `boot.loader.raspberryPi.useGenerationDeviceTree' in module `/nix/store/szyjiswb0gfzfss63m1p3a29zqf8kxzn-source/modules/system/boot/loader/raspberrypi'
 - option(s) with prefix `boot.loader.raspberryPi.variant' in module `/nix/store/szyjiswb0gfzfss63m1p3a29zqf8kxzn-source/modules/system/boot/loader/raspberrypi'
```

Shouldn't this issue be solved by disabling the remove module? Is there probably an issue with loading the modules in the correct order?
I'm using home-manager as well.

mrsteakhouse, 2025-02-27:
I got it to evaluate by disabling the whole removed file.

```nix
disabledModules = [
    (modulesPath + "/rename.nix")
  ];
```

How exactly can i debug this, to find the correct key to disable the module alone?
The former build command was incorrect, as the module does not supply the `sdImage` target.

nvmd, 2025-02-27:
Hi @mrsteakhouse, you're on the right way. The error you had was caused by the fact that `boot.loader.raspberryPi` was "removed" using `mkRemovedOptionModule` in the module you've disabled now. 

Unfortunately, with the current implementation in `nixpkgs` it's not possible to disable individual `mkRemovedOptionModule` and `doRename` because they create "anonymous modules" while not specifying `key` attribute. So you can't find anywhere, it is not automagically generated somehow, for example.

This is the reason why this flake uses not the upstream `nixpkgs`, but special fork [modules-with-keys](https://github.com/nvmd/nixpkgs/compare/modules-with-keys) there I've made the necessary changes. I believe it must be upstreamed, I didn't yet had time to try that though.
Individual commits: nvmd/nixpkgs@e31552238dce9af12e2f6cac14b9959aeb174c5e, nvmd/nixpkgs@c6442f3b5ab860072e6ef588b9c2fa782a98f2b9

This flake disables the `mkRemovedOptionModule` here:
https://github.com/nvmd/nixos-raspberrypi/blob/bcc388a12abb2fa16486d8d0172565901f9f2ed1/modules/system/boot/loader/raspberrypi/default.nix#L71

nvmd, 2025-02-27:
As for the order of the modules: no, but changing the order of the overlays may sometimes cause something unexpected.

nvmd, 2025-04-07:
Closing the issue as it appear to be solved with at least 2 possible ways:)

Feel free to open create a new issue if you have further questions.

closed 2025-04-07 by nvmd