nvmd/nixos-raspberrypi #4: How to build SD image?

poperigby, 2025-03-23:
I'm trying to build an SD image for my Raspberry Pi Zero 2 using this flake, but I'm not really sure how to. I have the following configuration for my system:

```nix
{ modulesPath, inputs, ... }:

{
  imports = [
    inputs.nixos-raspberrypi.nixosModules.sd-image
    inputs.nixos-raspberrypi.nixosModules.raspberry-pi-02.base
    inputs.nixos-raspberrypi.nixosModules.trusted-nix-caches

  disabledModules = [
    (modulesPath + "/rename.nix")
  ];

  time.timeZone = "America/Los_Angeles";

  services = {
    openssh.enable = true;
    octoprint.enable = true;
  };

  networking.networkmanager.enable = true;

  users = {
    users.nixos = {
      hashedPassword = "$y$j9T$t2WW7Q7yFNMpPerSi3FI.0$AALIY.Rd1IcVCS5lcZuZpLWdMCsOw9l7JLgGp4rYFPA";
      isNormalUser = true;
      group = "nixos";
      extraGroups = [ "wheel" ];
    };
    groups.nixos = { };
  };

  # https://wiki.nixos.org/wiki/FAQ/When_do_I_update_stateVersion
  system.stateVersion = "23.11";
}
```

Then I tried building it using this:

```
nix build .#nixosConfigurations.nestor.config.system.build.sdImage
```

I ended up with this error:

```
error:
       … while evaluating 'strict' to select 'drvPath' on it
         at /builtin/derivation.nix:1:552:
       … while calling the 'derivationStrict' builtin
         at /builtin/derivation.nix:1:208:
       (stack trace truncated; use '--show-trace' to show the full trace)

       error: attribute 'configuration-revision' missing
       at /nix/store/da0rc3kkzgs3bjybylwnb9qvp6gxym1n-source/modules/installer/sd-card/sd-image-raspberrypi.nix:15:39:
           14|   sdImage = {
           15|     imageBaseName = "nixos-sd-image-${config.configuration-revision.revision.short}";
             |                                       ^
           16|     firmwareSize = 1024;
```

nvmd, 2025-03-24:
Hi @poperigby,

The error was due to the missing module, I've fixed that now.

As for the usage – I've added an SD image installer for Zero2, which also doubles as a configuration example to `flake.nix`, and updated the README.
Take a look at `nixosConfigurations.rpi02-installer`. You can basically remove the "nixos' standard installer configuration" from it, and start adding your own stuff.

poperigby, 2025-03-24:
Thanks for fixing it so quickly! I'm getting a little confused with all the different modules, and I'm running into this error now:

```
error:
       … while evaluating 'strict' to select 'drvPath' on it
         at /builtin/derivation.nix:1:552:
       … while calling the 'derivationStrict' builtin
         at /builtin/derivation.nix:1:208:
       (stack trace truncated; use '--show-trace' to show the full trace)

       error: evaluation aborted with the following error message: 'cannot find attribute 'environment.checkConfigurationOptions''
```

I have these imports:

```
nixos-raspberrypi.lib.inject-overlays
nixos-raspberrypi.nixosModules.raspberry-pi-02.base
nixos-raspberrypi.nixosModules.nixpkgs-rpi
nixos-raspberrypi.nixosModules.sd-image-uboot
nixos-raspberrypi.nixosModules.trusted-nix-caches
```

nvmd, 2025-03-24:
You're welcome.
The imports are correct. The error is actually caused by
```
  disabledModules = [
    (modulesPath + "/rename.nix")
  ];
```
in your configuration. It disabled the following, which caused the error you have:
https://github.com/NixOS/nixpkgs/blob/f3a2a0601e9669a6e38af25b46ce6c4563bcb6da/nixos/modules/rename.nix#L21

To fix the issue, you could either keep the `nixpkgs` that this flake provides (it's quite recent `nixos-unstable`), or you could try adding the following "back", while keeping the `disabledModules`
```
(mkAliasOptionModuleMD [ "environment" "checkConfigurationOptions" ] [ "_module" "check" ])
```

poperigby, 2025-03-26:
@nvmd is there a way to use raspberrypi-utils without using your overlays? It's causing everything to rebuild and for some reason not hitting the cache.

nvmd, 2025-03-26:
Hi, how do you use the flake exactly?
It should hit the cache with this config.
```
inputs = {
  nixos-raspberrypi = {
    url = "github:nvmd/nixos-raspberrypi";
  };
};
```
If `nixpkgs.follows` is specified, then no guarantees, unfortunately.

Did you apply "global overlay"? "Everything" shouldn't even attempt to rebuild if you don't include it.
`raspberry-utils` is in the `vendor-pkgs` overlay.

nvmd, 2025-03-26:
One more thing about the cache. It came to me now, that it's not clear from the README: for the `nix build` to hit the cache you have them enabled on the system you're building as well.

For that,  either `trusted-nix-caches` have to be included in your local configuration, or the caches have to be enabled at least in the flake containing your raspberry system like this

```flake.nix
nixConfig = {
  extra-substituters = [
    "https://nixos-raspberrypi.cachix.org"
  ];
  extra-trusted-public-keys = [
    "nixos-raspberrypi.cachix.org-1:4iMO9LXa8BqhU+Rpg6LQKiGa2lsNh/j2oiYLNOQ5sPI="
  ];
};
```

poperigby, 2025-03-26:
No, I don't think I'm using the global overlay. I also don't have any `follows` set. These are my imports for my system:

```
nixos-raspberrypi.lib.inject-overlays
nixos-raspberrypi.nixosModules.raspberry-pi-02.base
nixos-raspberrypi.nixosModules.nixpkgs-rpi
nixos-raspberrypi.nixosModules.sd-image-uboot
nixos-raspberrypi.nixosModules.trusted-nix-caches
```

poperigby, 2025-03-27:
Here's my system configuration, if that's useful: https://codeberg.org/PopeRigby/nixos/src/branch/main/systems/aarch64-linux/nestor

nvmd, 2025-03-28:
Hi @poperigby, I've taken a look at the configuration you have on codeberg.

I've never worked with snowfall-lib before, but from what I see in the documentation, you seem to be right about the lib. Snowfall implicitly uses `nixpkgs` input to instantiate `nixosConfigurations` for all systems, but nixpkgs from `nixos-rapsberrypi` needs to be used. That also explains why you had to have the `disabledModules` in your config.
Snowflake should call the `<nixpkgs>.lib.nixosSystem` function somewhere to create the system configuration, I believe it should also allow overriding that default choice of `<nixpkgs>` somehow.
That's one thing. The other thing is, how you have caches enabled only in `nestor`s config, see my previous comment: https://github.com/nvmd/nixos-raspberrypi/issues/4#issuecomment-2755669245

nvmd, 2025-04-07:
Closing this support issue as the remaining question seems to be entirely dependent on the snowfall lib usage, which is out of scope of this project.
Feel free to open a new one, if you have further questions.

closed 2025-04-07 by nvmd