nvmd/nixos-raspberrypi #6: Infinite recustion encountered

LonguCodes, 2025-04-12:
Hi

I'm getting an error about recursion when using your module

```
error: infinite recursion encountered
       at /nix/store/1dwky0bis4bkl3qngsc6pmq902swa9b6-source/lib/modules.nix:620:66:
          619|       extraArgs = mapAttrs (
          620|         name: _: addErrorContext (context name) (args.${name} or config._module.args.${name})
             |                                                                  ^
          621|       ) (functionArgs f);
```

nvmd, 2025-04-12:
Hi @LonguCodes, could you please provide more information, which modules and overlays are you using and how exactly?

LonguCodes, 2025-04-12:
My full config

```
{
description = "Rpi";
inputs = {
  nixpkgs.url = "github:NixOS/nixpkgs/nixos-24.11";
  nixos-raspberrypi = {
    url = "github:nvmd/nixos-raspberrypi";
  };
};

outputs = { self, nixpkgs, ... }:
  let
    inherit (nixpkgs.lib) nixosSystem;
    basic-config = { pkgs, lib, ... }: {
      time.timeZone = "Europe/Warsaw";
      users.users.pi = {
        isNormalUser = true;
        extraGroups = [ "wheel" ];
        openssh.authorizedKeys.keys = [
          # SENSITIVE
        ];
      };

      environment.systemPackages = with pkgs; [
        emacs
        git
        wget
	neovim
      ];
      services.openssh = {
        enable = true;
      };
      system.stateVersion = "24.05";
    };

  in
    {
      nixosConfigurations = {
        rpi = nixosSystem {
          system = "aarch64-linux";
          modules = [ 
             ({ config, nixos-raspberrypi, lib, ... }: {
              imports = [
                # Hardware configuration
                nixos-raspberrypi.nixosModules.raspberry-pi-5.base

                # nixos-raspberrypi.lib.inject-overlays
                # nixos-raspberrypi.nixosModules.nixpkgs-rpi
              ];

              sdImage.imageBaseName = "nixos-sd-image";
            })
            basic-config
          ];
        };
      };
    };
}
```

nvmd, 2025-04-12:
You've missed `specialArgs = inputs` in `nixosSystem` call. That should solve it.
A couple more comments which you may find useful:
- `nixos-raspberrypi.lib.inject-overlays` is actually required now, it doesn't override any of the existing packages, just add the "missing" ones
- To use `sdImage` for RPi5 you need to import `sd-image-kernelboot`
- If you intend to use binary cache, use `lib.nixosSystem` from its `nixpkgs`, for example via `self.inputs.nixos-raspberrypi.inputs.nixpkgs`

LonguCodes, 2025-04-13:
The `specialArgs` helped, but now ii'm getting a different error

```
 The option `boot.loader.raspberryPi' in module `/nix/store/1dwky0bis4bkl3qngsc6pmq902swa9b6-source/nixos/modules/rename.nix' would be a parent of the following options, but its type `<no description>' does not support nested options.
        - option(s) with prefix `boot.loader.raspberryPi.bootPath' in module `/nix/store/ls7n42f4rxh5fzd9cd334h9z4w58zbhb-source/modules/system/boot/loader/raspberrypi'
        - option(s) with prefix `boot.loader.raspberryPi.bootPopulateCmd' in module `/nix/store/ls7n42f4rxh5fzd9cd334h9z4w58zbhb-source/modules/system/boot/loader/raspberrypi'
        - option(s) with prefix `boot.loader.raspberryPi.bootloader' in module `/nix/store/ls7n42f4rxh5fzd9cd334h9z4w58zbhb-source/modules/system/boot/loader/raspberrypi'
        - option(s) with prefix `boot.loader.raspberryPi.enable' in module `/nix/store/ls7n42f4rxh5fzd9cd334h9z4w58zbhb-source/modules/system/boot/loader/raspberrypi'
        - option(s) with prefix `boot.loader.raspberryPi.firmwarePackage' in module `/nix/store/ls7n42f4rxh5fzd9cd334h9z4w58zbhb-source/modules/system/boot/loader/raspberrypi'
        - option(s) with prefix `boot.loader.raspberryPi.firmwarePath' in module `/nix/store/ls7n42f4rxh5fzd9cd334h9z4w58zbhb-source/modules/system/boot/loader/raspberrypi'
        - option(s) with prefix `boot.loader.raspberryPi.firmwarePopulateCmd' in module `/nix/store/ls7n42f4rxh5fzd9cd334h9z4w58zbhb-source/modules/system/boot/loader/raspberrypi'
        - option(s) with prefix `boot.loader.raspberryPi.ubootPackage' in module `/nix/store/ls7n42f4rxh5fzd9cd334h9z4w58zbhb-source/modules/system/boot/loader/raspberrypi'
        - option(s) with prefix `boot.loader.raspberryPi.useGenerationDeviceTree' in module `/nix/store/ls7n42f4rxh5fzd9cd334h9z4w58zbhb-source/modules/system/boot/loader/raspberrypi'
        - option(s) with prefix `boot.loader.raspberryPi.variant' in module `/nix/store/ls7n42f4rxh5fzd9cd334h9z4w58zbhb-source/modules/system/boot/loader/raspberrypi'
```

nvmd, 2025-04-14:
Hi @LonguCodes, that's actually one more thing I forgot to mention.
Short answer: for the time being, you need to use `nixpkgs` provided by the flake, or you'll need to implement a workaround mentioned here (with more background as well):
- https://github.com/nvmd/nixos-raspberrypi/issues/4#issuecomment-2749544914
- https://github.com/nvmd/nixos-raspberrypi/issues/3#issuecomment-2689227585

This won't be necessary anymore once PR nixos/nixpkgs#398456 get's merged into `nixos-unstable`. You may want to 👍 that, maybe that can possibly speed up the process.

LonguCodes, 2025-04-15:
Yep, that helped (mostly). Thank you!

closed 2025-04-15 by LonguCodes