nvmd/nixos-raspberrypi #7: Trying to build an image for RPI5

harbinger0x7c0, 2025-04-13:
I'm not very familiar with flakes, so this may well be my misunderstanding. As near as I can tell if I want an SD image for a raspberry pi 5 I can customize the example installer.

After taking a crack at customization, I'm getting a compile error:
`error: The option 'boot.loader.generic-extlinux-compatible.populateCmd' was accessed but has no value defined.`

Did I do something to cause this field to get cleared? If not, how do I figure out what should go here?

[flake.nix.txt](https://github.com/user-attachments/files/19722471/flake.nix.txt)

nvmd, 2025-04-14:
Hi @harbinger0x7c0, I've updated installer infrastructure, adding dedicated installer configurations for Pi4 and Pi5 among others.
You should check that out, it should be a little easier to follow now. Your customizations could go into `custom-user-config`, for example.
Going further with configuration, you probably shouldn't need to modify this flake, but to consume it as one of the `inputs` of your own flake, borrowing the configuration you need from `nixosConfigurations` (as there isn't yet concise usage examples for now)

harbinger0x7c0, 2025-05-11:
Looks like the correct approach is different enough that the original request is a moot point. I'll close this and open another request that's more on-target if needed. Thanks!

closed 2025-05-11 by harbinger0x7c0