nvmd/nixos-raspberrypi #8: Using nixos-anywhere on the SD installer image fails

TymanWasTaken, 2025-04-14:
Hi, I am attempting to use this flake to install NixOS on my RPI Zero 2 W.

I am trying to use your SD installer image (extended with my SSH key) and then nixos-anywhere, but when I attempt to install my configuration, disko fails:
<details>
<summary>nixos-anywhere command</summary>

```sh
nix run github:nix-community/nixos-anywhere -- \
        --flake ".#rpi1" \
        --generate-hardware-config nixos-generate-config ./rpi-nix/hardware-configuration.nix \
        --extra-files ./nixos-anywhere-extras \
        --target-host "root@10.0.0.2"
```

</details>

<details>
<summary>disko output</summary>

```
### Formatting hard drive with disko ###
Warning: Permanently added '10.0.0.2' (ED25519) to the list of known hosts.
umount: /mnt: not mounted
++ realpath /dev/mmcblk0
+ disk=/dev/mmcblk0
+ lsblk -a -f
NAME        FSTYPE FSVER LABEL    UUID                                 FSAVAIL FSUSE% MOUNTPOINTS
ram0
ram1
ram2
ram3
ram4
ram5
ram6
ram7
ram8
ram9
ram10
ram11
ram12
ram13
ram14
ram15
loop0
loop1
loop2
loop3
loop4
loop5
loop6
loop7
mmcblk0
├─mmcblk0p1 vfat   FAT32 FIRMWARE 2175-794E
└─mmcblk0p2 ext4   1.0   NIXOS_SD 44444444-4444-4444-8888-888888888888    9.9G    22% /nix/store
                                                                                      /
+ lsblk --output-all --json
+ bash -x
++ dirname /nix/store/fpwn44vygjj6bfn8s1jj9p8yh6jhfxni-disk-deactivate/disk-deactivate
+ jq -r -f /nix/store/fpwn44vygjj6bfn8s1jj9p8yh6jhfxni-disk-deactivate/zfs-swap-deactivate.jq
+ lsblk --output-all --json
+ bash -x
++ dirname /nix/store/fpwn44vygjj6bfn8s1jj9p8yh6jhfxni-disk-deactivate/disk-deactivate
+ jq -r --arg disk_to_clear /dev/mmcblk0 -f /nix/store/fpwn44vygjj6bfn8s1jj9p8yh6jhfxni-disk-deactivate/disk-deactivate.jq
+ set -fu
+ wipefs --all -f /dev/mmcblk0p1
/dev/mmcblk0p1: 8 bytes were erased at offset 0x00000052 (vfat): 46 41 54 33 32 20 20 20
/dev/mmcblk0p1: 1 byte was erased at offset 0x00000000 (vfat): eb
/dev/mmcblk0p1: 2 bytes were erased at offset 0x000001fe (vfat): 55 aa
+ umount -R /nix/store
umount: /nix/store: target is busy.
+ umount -R /
umount: /nix/store: target is busy.
+ wipefs --all -f /dev/mmcblk0p2
/dev/mmcblk0p2: 2 bytes were erased at offset 0x00000438 (ext4): 53 ef
++ type zdb
++ zdb -l /dev/mmcblk0
++ sed -nr 's/ +name: '\''(.*)'\''/\1/p'
WARNING: ignoring tunable zfs_arc_max (using 132789760 instead)
+ zpool=
+ [[ -n '' ]]
+ unset zpool
++ lsblk /dev/mmcblk0 -l -p -o type,name
++ awk 'match($1,"raid.*") {print $2}'
+ md_dev=
+ [[ -n '' ]]
+ wipefs --all -f /dev/mmcblk0
/dev/mmcblk0: 2 bytes were erased at offset 0x000001fe (dos): 55 aa
/dev/mmcblk0: 8 bytes were erased at offset 0x3b597fe00 (gpt): 45 46 49 20 50 41 52 54
+ dd if=/dev/zero of=/dev/mmcblk0 bs=440 count=1
1+0 records in
1+0 records out
440 bytes copied, 0.000456618 s, 964 kB/s
+ lsblk -a -f
NAME        FSTYPE FSVER LABEL UUID FSAVAIL FSUSE% MOUNTPOINTS
ram0
ram1
ram2
ram3
ram4
ram5
ram6
ram7
ram8
ram9
ram10
ram11
ram12
ram13
ram14
ram15
loop0
loop1
loop2
loop3
loop4
loop5
loop6
loop7
mmcblk0
├─mmcblk0p1
└─mmcblk0p2                            9.9G    22% /nix/store
                                                   /
++ mktemp -d
+ disko_devices_dir=/tmp/tmp.sKVo3VrGK2
+ trap 'rm -rf "$disko_devices_dir"' EXIT
+ mkdir -p /tmp/tmp.sKVo3VrGK2
+ device=/dev/mmcblk0
+ imageName=main
+ imageSize=2G
+ name=main
+ type=disk
+ device=/dev/mmcblk0
+ efiGptPartitionFirst=1
+ type=gpt
+ blkid /dev/mmcblk0
+ sgdisk --clear /dev/mmcblk0
Creating new GPT entries in memory.
Warning: The kernel is still using the old partition table.
The new table will be used at the next reboot or after you
run partprobe(8) or kpartx(8)
The operation has completed successfully.
+ sgdisk --align-end --new=1:0:+1G --partition-guid=1:R --change-name=1:disk-main-ESP --typecode=1:EF00 /dev/mmcblk0
Warning: The kernel is still using the old partition table.
The new table will be used at the next reboot or after you
run partprobe(8) or kpartx(8)
The operation has completed successfully.
+ partprobe /dev/mmcblk0
Error: Partition(s) 2 on /dev/mmcblk0 have been written, but we have been unable to inform the kernel of the change, probably because it/they are in use.  As a result, the old partition(s) will remain in use.  You should reboot now before making further changes.
+ :
+ udevadm trigger --subsystem-match=block
+ udevadm settle --timeout 120
+ sgdisk --align-end --new=2:0:-0 --partition-guid=2:R --change-name=2:disk-main-root --typecode=2:8300 /dev/mmcblk0
Warning: The kernel is still using the old partition table.
The new table will be used at the next reboot or after you
run partprobe(8) or kpartx(8)
The operation has completed successfully.
+ partprobe /dev/mmcblk0
Error: Partition(s) 2 on /dev/mmcblk0 have been written, but we have been unable to inform the kernel of the change, probably because it/they are in use.  As a result, the old partition(s) will remain in use.  You should reboot now before making further changes.
+ :
+ udevadm trigger --subsystem-match=block
+ udevadm settle --timeout 120
+ device=/dev/disk/by-partlabel/disk-main-ESP
+ extraArgs=()
+ declare -a extraArgs
+ format=vfat
+ mountOptions=('umask=0077')
+ declare -a mountOptions
+ mountpoint=/boot
+ type=filesystem
+ blkid /dev/disk/by-partlabel/disk-main-ESP
+ grep -q TYPE=
+ mkfs.vfat /dev/disk/by-partlabel/disk-main-ESP
mkfs.fat 4.2 (2021-01-31)
+ device=/dev/disk/by-partlabel/disk-main-root
+ extraArgs=()
+ declare -a extraArgs
+ format=ext4
+ mountOptions=('defaults')
+ declare -a mountOptions
+ mountpoint=/
+ type=filesystem
+ blkid /dev/disk/by-partlabel/disk-main-root
+ grep -q TYPE=
+ mkfs.ext4 /dev/disk/by-partlabel/disk-main-root
mke2fs 1.47.2 (1-Jan-2025)
The file /dev/disk/by-partlabel/disk-main-root does not exist and no size was specified.
+ rm -rf /tmp/tmp.sKVo3VrGK2
Connection to 10.0.0.2 closed.
```

</details>

<details>
<summary>disko configuration</summary>

```nix
{
  disko.devices = {
    disk = {
      main = {
        type = "disk";
        device = "/dev/mmcblk0";
        content = {
          type = "gpt";
          partitions = {
            ESP = {
              size = "1G";
              type = "EF00";
              content = {
                type = "filesystem";
                format = "vfat";
                mountpoint = "/boot";
                mountOptions = [ "umask=0077" ];
              };
            };
            root = {
              size = "100%";
              content = {
                type = "filesystem";
                format = "ext4";
                mountpoint = "/";
              };
            };
          };
        };
      };
    };
  };
}
```

</details>

Reading the disko output, I believe this is likely the problem:
> Error: Partition(s) 2 on /dev/mmcblk0 have been written, but we have been unable to inform the kernel of the change, probably because it/they are in use.  As a result, the old partition(s) will remain in use.  You should reboot now before making further changes.

From my understanding, as the SD card is actively mounted and in use (for the nixos installer image), the kernel refuses to acknowledge the new partitioning and thus disko believes it failed. There doesn't seem to be a real example for utilizing disko for an RPI with this flake, so I assume this is just something I personally overlooked/misunderstood. How are disko/nixos-anywhere and this flake supposed to be used?

nvmd, 2025-04-15:
Hi @TymanWasTaken.
You're right, disko is unable to change the partitions and file systems because they're indeed in use by the current system, it also couldn't unmount `/` because it needs `/nix/store`.

The "disko/nixos-anywhere" scenario is primarily intended for RPi4/5 because it needs two storage devices:
- one with the running system. Network-booted system could also work, I guess, but I have no experience with that,
- the other being the target device.

It's easily achievable with RPi4/5 but not so much with RPi0.

As for RPi02 situation in this light:
- You could try booting it off the USB drive connected with USB hub. I haven't tried it, not sure if it's supported.
- Deploy the configuration you want directly to the installer with `nixos-rebuild` (is possible because unlike installer ISOs, SD-image is mutable).
From your disko configuration I see the partition layout you want is actually the same as `sd-image` uses, you could then just use this instead, you don't need to repartitioning:
```nix
          fileSystems = {
            "/boot/firmware" = {
              device = "/dev/disk/by-label/FIRMWARE";
              fsType = "vfat";
              options = [
                "umask=0077"
                "noatime"
                "noauto"
                "x-systemd.automount"
                "x-systemd.idle-timeout=1min"
              ];
            };
            "/" = {
              device = "/dev/disk/by-label/NIXOS_SD";
              fsType = "ext4";
              options = [ "noatime" ];
            };
          };
```

tobiasBora, 2025-04-21:
I'm also confused by this… I have a rpi5, but no nvme disk, just a regular SD card, and I want to use a zfs partition. What do you recommend here?

nvmd, 2025-04-21:
Hi!
Yes, booting from USB would be my suggestion, yes.
As for the `kexec`, you're also right, `nixos-anywhere`, for example, makes use of that on other platforms. It didn't work out of the box on RPi5 the last time I tried.
Support of `kexec`-installation is something that wasn't yet tested or worked on in this project.

tobiasBora, 2025-04-22:
Ok, so I can confirm that booting & installing from USB works (just removed the SD card, plugged the USB, booted, put the SD card back and ran the installation): now I run NixOS on ZFS! :-D Thanks a lot, maybe this should be documented in the README if `kexec` is not yet supported/tested as I was quite confused when I read the doc first.

nvmd, 2025-04-24:
Now that I know it can cause confusion, I'll absolutely add a note to the README. Thanks for the feedback!

As for booting and installation, you could also edit boot order with `rpi-eeprom-config` (should be available in installer image), so that RPi always tries booting from USB first. That could save you a couple of pluggin-unplugging procedures.  See https://www.raspberrypi.com/documentation/computers/raspberry-pi.html#BOOT_ORDER

tobiasBora, 2025-04-24:
Awesome, thanks!

TymanWasTaken, 2025-04-27:
I can confirm that skipping the nixos-anywhere and directly booting off of the `sd-image.nix` configuration works for me, no usb-boot necessary.

A note in the readme about when nixos-anywhere is supposed to be used for other users would be great, when you get the time :)

Thanks for making this project!

closed 2025-05-04 by TymanWasTaken