nvmd/nixos-raspberrypi #11: Cache nixos installer

tobiasBora, 2025-04-20:
To install this great looking project, I just tried to clone this repo and build the installer (after adding my ssh key, is it how we should do that?):
```
nix build .#installerImages.rpi5 
```
However it seems that this is not part of the build cache as it tries to build ` linux-config-6.6.74-stable_20250127`, `linux_rpi-bcm2712-6.6.74-stable_20250127` and others. And building a linux aarch64 kernel on a x86_64 will probably take ages… Any idea why I don't get the cache? I did answered to the questions at the beginning:
```
do you want to allow configuration setting 'connect-timeout' to be set to '5' (y/N)? y
do you want to permanently mark this value as trusted (y/N)? N
do you want to allow configuration setting 'extra-substituters' to be set to 'https://nixos-raspberrypi.cachix.org' (y/N)? y
do you want to permanently mark this value as trusted (y/N)? N
do you want to allow configuration setting 'extra-trusted-public-keys' to be set to 'nixos-raspberrypi.cachix.org-1:4iMO9LXa8BqhU+Rpg6LQKiGa2lsNh/j2oiYLNOQ5sPI=' (y/N)? y
do you want to permanently mark this value as trusted (y/N)? N
```

nvmd, 2025-04-21:
Hi @tobiasBora.
Is your user a trusted one on your current system (i.e. added to [`nix.settings.trusted-users`](https://search.nixos.org/options?channel=unstable&from=0&size=50&sort=relevance&type=packages&query=trusted-users))?

nvmd, 2025-04-24:
Yes, sure, makes perfect sense. There's also `nixos-raspberrypi.nixosModules.trusted-nix-caches` which could be imported on both systems (i.e. your builder and raspberry), so that you wouldn't need to copy the public keys manually.
I got where you're going with the documentation improvement, thanks for the feedback!
If you have some ideas about particular wording, contributions are always welcome:-)