nvmd/nixos-raspberrypi #12: Need help with device tree overlays and cache usage

padd1er, 2025-04-24:
To be short, I had some issues with default and community flakes on my rpi4. So, I decided to try using this flake now. 
But seems I cannot get everything to work correctly - I am still somewhat new with Nix and NixOS. Any help would be appreciated.

This is the latest rebuild log `sudo nixos-rebuild --impure --flake . test |& nom`, for whatever reason it does not seems to use cache and trying to build everything locally, including `linux-config-6.6.74-stable_20250127` and `linux_rpi-bcm2711-6.6.74-stable_20250127`, which took over 8hrs and still failed at the end:

<details>
<summary>rebuild log</summary>

```
building '/nix/store/f5yirxbbj4119i4kbzkkrfcaj1d9wgar-linux_rpi-bcm2711-6.6.74-stable_20250127.drv'...
building '/nix/store/q8psrv61np7c5g0lwn5301vn274g5k8d-argonone-enable-overlay-dtbo.drv'...
building '/nix/store/0i5q35plvb3jyqc0517wvbafk5i7cs4h-linux_rpi-bcm2711-6.6.74-stable_20250127-modules.drv'...
building '/nix/store/7hv1pczb09k92bfa033vvmyj8g4yz03z-overlays.json.drv'...
building '/nix/store/n8gvdr029r26s4ql34ql3qmfv8n7vyj0-linux_rpi-bcm2711-6.6.74-stable_20250127-modules-shrunk.drv'...
building '/nix/store/6xl6xklnxfylpmiximw2yx7d97d2xsd5-device-tree-overlays.drv'...
error: builder for '/nix/store/6xl6xklnxfylpmiximw2yx7d97d2xsd5-device-tree-overlays.drv' failed with exit code 1;
       last 12 log lines:
       > Processing source device tree overlays/hat_map.dtb...
       >   Device tree overlays/hat_map.dtb has no compatible string set. Assuming it's compatible with overlay
       >   Applying overlay argonone
       > Traceback (most recent call last):
       >   File "/nix/store/3w8w3va9xc0qr91d849v25rh8iivhipv-apply_overlays.py", line 110, in <module>
       >     main()
       >   File "/nix/store/3w8w3va9xc0qr91d849v25rh8iivhipv-apply_overlays.py", line 100, in main
       >     dt = apply_overlay(dt, overlay.fdt)
       >          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
       >   File "/nix/store/3w8w3va9xc0qr91d849v25rh8iivhipv-apply_overlays.py", line 57, in apply_overlay
       >     raise FdtException(err)
       > libfdt.FdtException: pylibfdt error -4: FDT_ERR_BADOFFSET
       For full logs, run:
         nix log /nix/store/6xl6xklnxfylpmiximw2yx7d97d2xsd5-device-tree-overlays.drv
error: 1 dependencies of derivation '/nix/store/6c79b8ip0jhmfdqmnnn7nd9adjv92ymz-nixos-system-box-nixos-25.05.20250421.c11863f.drv' failed to build
┏━ Dependency Graph:
┃    ┌─ ✔ uboot-rpi_arm64_defconfig-2025.01 ⏱ 7m10s
┃ ┌─ ✔ uboot-builder.sh ⏱ 1s
┃ │  ┌─ ✔ system_fish-completions ⏱ 4m38s
┃ │  │     ┌─ ✔ X-Restart-Triggers-polkit ⏱ 1s
┃ │  │  ┌─ ✔ unit-polkit.service ⏱ 14s
┃ │  │  ├─ ✔ unit-dbus.service ⏱ 2s
┃ │  ├─ ✔ system-units ⏱ 1m52s
┃ │  │        ┌─ ✔ dbus-1 ⏱ 1s
┃ │  │     ┌─ ✔ X-Restart-Triggers-dbus ⏱ 7s
┃ │  │  ┌─ ✔ unit-dbus.service ⏱ 2s
┃ │  ├─ ✔ user-units ⏱ 1m53s
┃ │  │  ┌─ ✔ man-cache ⏱ 5m55s
┃ │  ├─ ✔ etc-man_db.conf ⏱ 1s
┃ ├─ ✔ etc ⏱ 26s
┃ │     ┌─ ⏸ closure-info waiting for 1 ⏵
┃ │     │  ┌─ ✔ linux_rpi-bcm2711-6.6.74-stable_20250127-modules ⏱ 2s
┃ │     ├─ ⏵ linux_rpi-bcm2711-6.6.74-stable_20250127-modules-shrunk ⏱ 2s
┃ │  ┌─ ⏸ initrd-linux_rpi-bcm2711-6.6.74-stable_20250127
┃ ├─ ⏸ boot.json
┃ │              ┌─ ✔ source ⏱ 8m35s
┃ │           ┌─ ✔ linux-config-6.6.74-stable_20250127 ⏱ 2m39s
┃ │        ┌─ ✔ linux_rpi-bcm2711-6.6.74-stable_20250127 ⏱ 8h15m9s
┃ │     ┌─ ✔ argonone-enable-overlay-dtbo ⏱ 1s
┃ │  ┌─ ✔ overlays.json ⏱ 2s
┃ ├─ ⚠ device-tree-overlays failed with exit code 1 after ⏱ 1s
┃ ⏸ nixos-system-box-nixos-25.05.20250421.c11863f
┣━━━ Builds            │ Downloads         │ Host
┃    ⏵ 1 │ ✔ 148 │     │     │       │     │ localhost
┃        │       │     │     │ ↓ 110 │     │ [1]: https://cache.nixos.org
┃        │       │     │     │ ↓   2 │     │ [2]: https://nix-community.cachix.org
┗━ ∑ ⏵ 1 │ ✔ 148 │ ⏸ 4 │ ↓ 0 │ ↓ 112 │ ⏸ 0 │ ⚠ Exited after 1 build failures at 08:23:09 after 8h28m12s
```

</details>

My stripped configuration below:
<details>
[... 212 lines cut ...]

  disabledModules = [
    (modulesPath + "/rename.nix")
  ];

  fileSystems."/" = {
    device = "/dev/disk/by-uuid/44444444-4444-4444-8888-888888888888";
    fsType = "ext4";
  };

  swapDevices = [ ];

  # Enables DHCP on each ethernet and wireless interface. In case of scripted networking
  # (the default) this is the recommended approach. When using systemd-networkd it's
  # still possible to use this option, but it's recommended to use it in conjunction
  # with explicit per-interface declarations with `networking.interfaces.<interface>.useDHCP`.
  networking.useDHCP = lib.mkDefault true;
  # networking.interfaces.end0.useDHCP = lib.mkDefault true;
  # networking.interfaces.wlan0.useDHCP = lib.mkDefault true;

  nixpkgs.hostPlatform = lib.mkDefault "aarch64-linux";
}
```

</details>

nvmd, 2025-04-24:
Hi!

## ArgonOne
I've tried to fix `argonone` flake sometime ago, but its overlay is actually still "broken" – that's the reason your build failed. I'll mark its module as such.
You can try configuring in the way upstream recommends: copying its provided overlays, configure it in `config.txt` (i.e. `hardware.raspberry-pi.config`) and maybe use the daemon from the `argonone` flake.

## Caches
There're a couple of problems here:
1. You're making `nixos-raspberrypi ` follow your `inputs.nixpkgs`. I build the binary cache only for the `nixpkgs` locked by this flake. That means, that sometimes binary cache may work for you, and sometimes not.
```
    nixos-raspberrypi = {
      url = "github:nvmd/nixos-raspberrypi";
      inputs.nixpkgs.follows = "nixpkgs"; # This line
    };
```
2. The caches are probably not configured correctly: `cacheModule` you have in your flake applies only to the systems you're building, not the local one you're building them **on**. You either can:
   * Add those caches to `nixConfig` of the flake, like in [REAME](https://github.com/nvmd/nixos-raspberrypi/tree/main?tab=readme-ov-file#add-flake-input). Your local user must be trusted for this to work (see [nix.settings.trusted-users](https://search.nixos.org/options?channel=unstable&type=packages&query=trusted-users))
   * Add those caches to the local configuration first, and then build the configurations in your flake. For `nixos-raspberrypi` you can import`nixos-raspberrypi.nixosModules.trusted-nix-caches` both in your local and in Raspberry configurations.

padd1er, 2025-04-25:
Thanks a lot for the explanation!
I'll try to workaround those, now at least I understand what is happening.

closed 2025-04-25 by padd1er