nvmd/nixos-raspberrypi #13: Do overlays like for Hifiberry DACs work?

Cameo007, 2025-04-25:
Maybe this question is unnecessary but before I flash my Pi 5 I would like to know whether overlays work completely. I mean the Pi does not only need the line in `config.txt` like this
```
dtoverlay=hifiberry-dacplus-std
```
but also the drivers, am I right?

nvmd, 2025-04-25:
Hi!
This flake uses the latest stable Raspberry's vendor kernel and firmware by default, so everything should work like it works in their official distribution, configurable with `config.txt` (i.e. `hardware.raspberry-pi.config`).

As for specifically Hifiberry DAC, I have no experience with it. In case its driver doesn't come with the kernel and needs to be installed separately, it'll need to be packaged for NixOS first.

nvmd, 2025-05-20:
Closing this issue for now. Feel free to open a new issue or a discussion if you have further questions!

closed 2025-05-20 by nvmd