nvmd/nixos-raspberrypi #14: Struggling with encrypted root ZFS

tobiasBora, 2025-05-07:
Thanks for this cool project. I would like to have an encrypted root ZFS with the key present in `/boot/secret.txt` (usecase: zfs send an encrypted partition to an untrusted backup server). I managed to have ZFS without key, but I'm struggling to make it work with a key. By putting `echo` in various phases, I realized that it seems to hang (or at least I have no more outputs) between `preDeviceCommands` and `preLVMCommands`, i.e. around these lines https://github.com/NixOS/nixpkgs/blob/5b35d248e9206c1f3baf8de6a7683fee126364aa/nixos/modules/system/boot/stage-1-init.sh#L270-L282:

```
systemd-udevd --daemon
udevadm trigger --action=add
udevadm settle
```

Which makes sense since the last message I see relates to systemd. Sadly, `preDeviceCommands` is too early to mount stuff (I get errors like can't lookup blockdev). The last message I see is:

```
running udev...
Starting systemd-udevd version 257.3
[ 1.721181] macb 1f001000000.ethernet end0: renamed from eth0
[ 1.762017] vc4-drm axi:gpu: bcm2712_iommu_of_xlate: MMU 1000005200.iommu ## <-- this line is not always present
```

Any idea why the system hangs here? I've not even the time to run my script that mounts `/boot` in preLVMCommands.

Here is the flake I use:
```
{
  description = "My raspberry pi flake (nvmd/raspberrypi version)";

  # Optional: Binary cache for the flake
  nixConfig = {
    extra-substituters = [
      "https://nixos-raspberrypi.cachix.org"
    ];
    extra-trusted-public-keys = [
      "nixos-raspberrypi.cachix.org-1:4iMO9LXa8BqhU+Rpg6LQKiGa2lsNh/j2oiYLNOQ5sPI="
    ];
  };
  inputs = {
    nixpkgs.url = "github:NixOS/nixpkgs/nixos-24.11";
    nixos-raspberrypi.url = "github:nvmd/nixos-raspberrypi";
    # Format sd card
    # disko.url = "github:nix-community/disko/latest";
    # disko.inputs.nixpkgs.follows = "nixpkgs";
    disko = {
      # the fork is needed for partition attributes support
      url = "github:nvmd/disko/gpt-attrs";
      # url = "github:nix-community/disko";
      inputs.nixpkgs.follows = "nixos-raspberrypi/nixpkgs";
    };
  };

  outputs = { self, nixpkgs, nixos-raspberrypi, disko, ... }@inputs:
    let
      inherit (nixpkgs.lib) nixosSystem;
      minimalist-config = { pkgs, lib, ... }: {
        time.timeZone = "Europe/Paris";
        boot.supportedFilesystems = [ "zfs" "vfat" ];
        #fileSystems."/boot".neededForBoot = true;
        ## During the stage1 boot, nix will automatically prepend /mnt-root to all mounted files
        ## as written in
        ## https://github.com/NixOS/nixpkgs/blob/5b35d248e9206c1f3baf8de6a7683fee126364aa/nixos/modules/system/boot/stage-1-init.sh#L537-L551
        ## before doing a switch-root at the very end. Hence, /boot will certainly
        ## also be mounted in /mnt-root/boot, but since the ZFS partition is root, ZFS will be mounted first. Hence, to avoid this issue, we do it in two steps: we manually mount the partition somewhere (not in /boot since I don't know if this already exists), copy the file to /boot/, and later unmount ourselve
        ## before ZFS (lib.mkBefore needed since zfs activation runs in postResumeCommands):
        boot.initrd.preDeviceCommands = lib.mkBefore ''
          echo "XXXXX in preDeviceCommands"
        '';
        boot.initrd.preLVMCommands = lib.mkBefore ''
          echo "XXXXX in preLVMCommands"
          ls /
          mkdir -p /boot
          ls /
          mount /dev/disk/by-partlabel/ESP /boot
          #espmountlocation=$(mktemp -d)
[... 237 lines cut ...]
                        };
                      };
                    };
                    
                  };
                };
                # Use uboot (kernelboot does not list all generations)
                boot.loader.raspberryPi.bootloader = "uboot";
                # Used to generate the name of the current generation
                system.nixos.tags = let
                  cfg = config.boot.loader.raspberryPi;
                in [
                  "raspberry-pi-${cfg.variant}"
                  cfg.bootloader
                  config.boot.kernelPackages.kernel.version
                ];
                system.stateVersion = "25.05";
              })
            ];
          };
        };
      };
}

```

tobiasBora, 2025-05-08:
So after using the kernel option `boot.trace`, it hangs on a line `udevadm settle`.

tobiasBora, 2025-05-09:
After adding `udev.log_level=debug` to kernel parameters:

[image]

I can see quite a few lines about missing modules (even if it is not the last one), like:
```
failed to find module 'of:NiommuT(null)Cbrcm,bcm2712-iommu
```
if it rings a bell.

tobiasBora, 2025-05-09:
Also found this related discussion https://hub.mender.io/t/boot-stopping-after-four-berries/7256/6 that mentions that they got it fixed by removing `dtoverlay=vc4-kms-v3d` from config.txt… No idea what this may produce (will gui don't work anymore?), I need to try.

nvmd, 2025-05-09:
Hi @tobiasBora, I see you're using `uboot`. As far as I know, it still has somewhat limited support, maybe that could affect the boot process. Did you try `kernelboot` with this configuration by any chance?

Did "failed to find module" lines appear in the log when you're not using zfs encryption? I definitely don't see any of those lines when booting with `kernelboot` and unencrypted zfs.

Ideally `vc4` shouldn't affect what you're doing, at least directly, that's gpu module and its respective configuration.

I personally never tried to configure encrypted zfs with keyfile unlock, but I did configure Luks/LVM. Maybe this module could be of any help: https://gist.github.com/nvmd/e46f6c299fc7c2a98e90084f750b8092.
Maybe `boot.initrd.kernelModules` and which `*Commands` are used.

tobiasBora, 2025-05-11:
@nvmd Thanks a lot for your answer, it allowed me to fix it with:
```
        boot.initrd.kernelModules = [
          "uas" "usbcore" "usb_storage"
          "vfat" "nls_cp437" "nls_iso8859_1"
          "ext4"  # in case ext4 is not configured as builtin
        ];
```

Now I can boot safely even with uboot! (Just, unrelated, but when uboot starts, I only see a small picture with uboot logo, but no text, and no way to select a different boot option with the keyboard (but works if I manually modify the extlinux file in /boot and reboot)… is it a bug or am I doing something stupid?)

And I realized that for some reasons, everything after udev's run is NOT printed on the HDMI, only visible in the journalctl logs (but I can only see when it succeeds, I don't know if I can access it when it fails…). I tried to record the screen and even frame-by-frame recording never shows all the text after udev's run, explaining why I thought it was failing during udev… I guess it was actually failing during the mount. Any idea why?

Otherwise, I guess this issue can be closed, unless you have a way to automatically detect that we need this code…

tobiasBora, 2025-05-11:
Oh, I forgot: when it boots, it hangs for 1mn30 on:
```
A start job is running for /sys/subsystem/net/devices/usb0 (1mn 16s / 1mn 30s)
```
Any idea why?

nvmd, 2025-05-11:
It seems to be waiting for USB Gadget/Ethernet device (usb0).
Do you have `usb-gadget-ethernet` imported? I guess an additional configuration is needed for it to function on rpi5.

—
SK

-------- Original Message --------

tobiasBora, 2025-05-12:
Oh good point, thanks! I don't need it for now, so I guess I will just disable it. Thanks!

closed 2025-05-12 by tobiasBora