nvmd/nixos-raspberrypi #15: Follow nixpkgs 24.11

ptrsr, 2025-05-10:
Hi, first of all thank you for this repo! It's been working really well so far.

The only thing I'm struggeling with is pinning the `nixpkgs` to a specific version, in my case `24.11`.

I have tried;
```
inputs = {
  nixpkgs.url = "github:NixOS/nixpkgs/nixos-24.11";
  nixos-raspberrypi = {
    url = "github:nvmd/nixos-raspberrypi";
    inputs.nixpkgs.follows = "nixpkgs";
  };
```
But that results in;
```
error: The option `boot.loader.raspberryPi' in module `/nix/store/lzml2qh553njgdalw5cbbc208vwachr6-source/nixos/modules/rename.nix' would be a parent of the following options, but its type `<no description>' does not support nested options.
```

The README does specify something about `nixpkgs` being an optional input, so far I've tried adding it like this;

```
nixos-raspberrypi.lib.nixosSystem {
  system      = "aarch64-linux";
  specialArgs = { nixos-raspberrypi; };
  nixpkgs = nixpkgs;
}
```
But that returns the following error;
```
      error: function 'anonymous lambda' called with unexpected argument 'nixpkgs'
       at /nix/store/lzml2qh553njgdalw5cbbc208vwachr6-source/nixos/lib/eval-config.nix:11:1:
           10| # types.submodule instead of using eval-config.nix
           11| evalConfigArgs@{
             | ^
```

Could you elaborate what the correct approach is?

bdew, 2025-05-16:
The problem is that this repo uses a modified version of nixpkgs unstable (https://github.com/nvmd/nixpkgs/tree/modules-with-keys-unstable)

There's an open issues in nixpkgs to integrate that change - https://github.com/NixOS/nixpkgs/pull/398456

If you try to build it with stock nixpkgs (stable or no) it will fail with that module removed error.

You could make your own fork and rebase that patch to 24.11, but you'd have to manually maintain that.

koalalorenzo, 2025-05-17:
I have had the same issue, and no solution to find. I even get this error when building:

```
error:
       … while calling the 'seq' builtin
         at /nix/store/ygmv1k0z09f925012svfmqxadmzv7rc4-source/lib/modules.nix:359:18:
          358|         options = checked options;
          359|         config = checked (removeAttrs config [ "_module" ]);
             |                  ^
          360|         _module = checked (config._module);

       … while evaluating a branch condition
         at /nix/store/ygmv1k0z09f925012svfmqxadmzv7rc4-source/lib/modules.nix:295:9:
          294|       checkUnmatched =
          295|         if config._module.check && config._module.freeformType == null && merged.unmatchedDefns != [ ] then
             |         ^
          296|           let

       (stack trace truncated; use '--show-trace' to show the full, detailed trace)

       error: undefined variable 'systemdUtils'
       at /nix/store/ygmv1k0z09f925012svfmqxadmzv7rc4-source/nixos/modules/system/boot/systemd.nix:10:6:
            9| with utils;
           10| with systemdUtils.unitOptions;
             |      ^
           11| with lib;
```

I am assuming that this is because nvmd/nixos-raspberrypi's using a old fork of nixpkgs-unstable (as mentioned).

I have left a thumb sup to https://github.com/NixOS/nixpkgs/pull/398456 , I hope it will help.

Is there any other way to use it?

nvmd, 2025-05-17:
Hi!

@bdew thank you for a correct and concise answer!

Speaking of the fork, I also have <https://github.com/nvmd/nixpkgs/tree/modules-with-keys-stable> which I update sporadically (I did so today).

@ptrsr you could try specifying `nixpkgs.url = "github:nvmd/nixpkgs/modules-with-keys-stable";`, maybe that could work you.
I'm unable to provide any binary caches or test any configurations based on that, though. However, as soon as 25.05 gets released, I think this flake will be pinned to nixos stable for some time.

@koalalorenzo I've updated `main` branch to the more actual version of `nixos-unstable`. Does it work for you?

When `main` looks outdated, you could also try `develop` branch. It's not broken _most_ of the time, just less tested. I'm doing occasional forced pushes there, though.

koalalorenzo, 2025-05-21:
> [@koalalorenzo](https://github.com/koalalorenzo) I've updated `main` branch to the more actual version of `nixos-unstable`. Does it work for you?
>
> When `main` looks outdated, you could also try [...]

Sadly no :(  I get the same error about `systemdUtils`!

koalalorenzo, 2025-06-03:
@nvmd I still get the same issue with 25.05 or nixpkgs-unstable :(

koalalorenzo, 2025-06-03:
@nvmd turnout that the culprit was:

```
    inputs.utils.url = "github:numtide/flake-utils";
```

Removing `flake-utils` fixed it, I can't still generate the image but at least the issue is different than `systemdUtils` missing ;-)

uhlajs, 2025-09-10:
@nvmd Is there any update regarding the [nixpkgs#398456](https://github.com/NixOS/nixpkgs/pull/398456). I can see that there are some suggestions by reviewer, but once this is resolved, it should be possible to merge that.