nvmd/nixos-raspberrypi #18: nixos-generators

koalalorenzo, 2025-05-17:
It would be super useful to see in the docs how to use `nixos-generators` to generate sd images :)

Does this project work with [nixos-generators](https://github.com/nix-community/nixos-generators)?

nvmd, 2025-05-17:
Hi @koalalorenzo! I don't have much experience with nixos-generators, so can't really tell you if it will or should work with it:)

I suppose some image-generation logic of the nixos-generators could be reused (imported) here, so we won't need to reinvent the wheel. This isn't something I looked into yet, though.

That's definitely something worth considering, thank you for an idea. Having said that, contributions are always welcome!:)

thelenlucas, 2025-05-19:
I think it's definitely possible - it's been a minute since I've looked at it, but we have a headless nixos-generators flake working for a Raspberry Pi 4B.

thelenlucas, 2025-05-19:
It seems like the image built here is a better version of [the nixpkgs version](https://github.com/NixOS/nixpkgs/blob/master/nixos/modules/installer/sd-card/sd-image-raspberrypi.nix), which was barely ever supported (a leftover of ARM support on nixos being so-so I think). NixOS-Generators seems to use the base /sd-image-aarch64.nix implementation, so at the minimum it should be possible to merge some of the functionality of them here (or even overlay our module onto theirs? I'm far form a nix wizard).