nvmd/nixos-raspberrypi #30: Failed to install bootloader on a Raspberry Pi 5's SD card

doggobit0, 2025-06-03:
I would really really appreciate some help, I've been banging my head for hours on this :). So I'm trying to deploy to my Raspberry Pi 5. I don't have an SD card, so I'm just trying to build on the SD card itself. It's probably overkill but I:
1. Built the RPi 5 image
2. Trying to deploy via nix-anywhere

My flake.nix:
```nix
{
	description = ":(((";
	inputs = {
		nixpkgs.url = "github:NixOS/nixpkgs/nixos-25.05";
		nixos-raspberrypi.url = "github:nvmd/nixos-raspberrypi";
		disko.url = "github:nix-community/disko";
	};

	nixConfig = {
		extra-substituters = [
			"https://nixos-raspberrypi.cachix.org"
		];
		extra-trusted-public-keys = [
			"nixos-raspberrypi.cachix.org-1:4iMO9LXa8BqhU+Rpg6LQKiGa2lsNh/j2oiYLNOQ5sPI="
		];
	};

	outputs = { self, nixpkgs, nixos-raspberrypi, disko }@inputs:
		{
			nixosConfigurations = {
				bright-reindeer = nixos-raspberrypi.lib.nixosSystem {
					specialArgs = inputs;
					modules = [
						disko.nixosModules.disko

						({...}: {
							imports = with nixos-raspberrypi.nixosModules; [
								raspberry-pi-5.base
								raspberry-pi-5.bluetooth
							];
						})

						({ ... }: {
							networking.hostName = "unicorn";

							users.users.corn = {
								initialPassword = "unicornuni";
								isNormalUser = true;
								extraGroups = [
									"wheel"
								];
							};

							services.openssh.enable = true;
						})

						({ ... }: {
							fileSystems = {
								"/boot/firmware" = {
									device = "/dev/disk/by-uuid/2175-794E";
									fsType = "vfat";
									options = [
										"noatime"
										"noauto"
										"x-systemd.automount"
										"x-systemd.idle-timeout=1min"
									];
								};
								"/" = {
									device = "/dev/disk/by-uuid/44444444-4444-4444-8888-888888888888";
									fsType = "ext4";
									options = [ "noatime" ];
								};
							};

							disko.devices = {
								disk = {};
							};
						})
					];
				};
			};
		};
}
```

Logs:
```
### Installing NixOS ###
Pseudo-terminal will not be allocated because stdin is not a terminal.
Warning: Permanently added '192.168.1.5' (ED25519) to the list of known hosts.
installing the boot loader...
setting up /etc...
kernelboot-builder: -f /boot/firmware -c /nix/store/64cjqa0y1gg3srrgs3d4ar5a2k0yvwl4-nixos-system-unicorn-25.05.20250505.1473ad1
copying firmware...
using vendor firmware from /nix/store/irvijwviy627zkvs87wd49jrk2cbs16q-raspberrypi-firmware-1.20250430/share/raspberrypi/boot
cp: cannot create regular file '/boot/firmware/bcm2708-rpi-b.dtb.tmp': No such file or directory
Failed to install bootloader
```

Command:
```
$ nixos-anywhere --flake .#unicorn root@192.168.1.5 --phases kexec,install,reboot --build-on remote
```

nvmd, 2025-06-03:
Hi @doggobit0. It's really hard to say right now, I don't quite understand what is you situation. Could you describe in more detail? For example, 
So do you already have an installed nixos on RPi5?
Where are you running `nixos-anywhere` from? I've also noticed that you're attempting to deploy `.#unicorn`, but the configuration you attached is for `bright-reindeer`.
Regarding the error... what's the size of `/boot/firmware/`? Maybe where's no free space?

nvmd, 2025-06-03:
> If instead of doing this I should be using the installer, that would be great. However, for some reason it seems to be struggling to "install" on the same disk it's running off of as well :(

Yes, this scenario and `kexec` isn't currently supported, unfortunately.

> 1. I built one of the SD installer images via `nix build github:nvmd/nix-raspberrypi#installerImages.rpi5`, put it on an SD card. That's what's currently running on my RPi 5.

Now, when you already have nixos running on RPi5 on sdcard (which is also your only disk device), you should be able to `nixos-rebuild switch ... --target-host ...`.
If you had one more drive connected, you could partition it and install nixos onto it with `nixos-anywhere`.

> Trying to do `nixos-generate-config` on the SD installer directly and then `nixos-rebuild switch` fails with this error instead:

Hm, this one is very interesting. Anyway, try executing `nixos-rebuild switch ... --target-host ...` from your other nixos host first. We'll get back to this latter error if needed.

nvmd, 2025-06-03:
> 3\. I tried to set up the flake to work with `nixos-anywhere`. This is the first time I'm working with that tool, so my knowledge is quite shaky. It complained that I had no disko setup, that's why I added these lines:

Well, in your case you don't actually need `disko.nixosModules.disko` and its empty config at all – no partitioning is needed and you already have your `fileSystems` configured.

nvmd, 2025-06-03:
> Trying to do `nixos-generate-config` on the SD installer directly and then `nixos-rebuild switch` fails with this error instead:

Just don't do `nixos-generate-config` :) Everything it could potentially do is actually pre-configured in board modules of this flake. I guess it's just trying to configure `boot.loader.grub` somehow, breaking the config.

doggobit0, 2025-06-03:
Welp, thanks for the help, simplest things are the most robust sometimes, I literally just built the flake directly on the device, and it worked perfectly :). At some point in the future I'll try to unpick why some of these things failed, but until then, I finally have a working system haha.

Thanks for the awesome project ^.^

closed 2025-06-03 by doggobit0

doggobit0, 2025-06-03:
@nvmd Sorry for necro'ing, but just wanted to give you a heads up. I modified the NixOS on ARM/Raspberry Pi 5 wiki page: https://wiki.nixos.org/wiki/NixOS_on_ARM/Raspberry_Pi_5

In case people come with broken configs, it might be the reason. Feel free to add/remove/modify anything there, I tested the config on my setup and seemed to work fine, but yknow, just in case.

nvmd, 2025-06-04:
@doggobit0 Absolutely no problem, thanks for a heads-up. This kind of easy-to-follow documentation is exactly what this flake needs – thank you so much for your effort!
Well, people a going to come with broken configs anyway, but maybe they'll come a little rarely... ;)

While we're still here, a small tip: I'm planning to change the default branch to `develop` soon, so use `github:nvmd/nixos-raspberrypi/main` for flake url if you want to make sure you stay on "stable".
I'm in the process of updating that in the docs.