nvmd/nixos-raspberrypi #34: Simple config does not work, options being defined by nixpkgs

schuelermine, 2025-06-11:
I have:
```nix
{
  description = "A very basic flake";

  inputs = {
    nixos-raspberrypi.url = "github:nvmd/nixos-raspberrypi/main";
    nixos-raspberrypi.inputs.nixpkgs.url = "github:NixOS/nixpkgs/nixos-24.11";
  };

  outputs = inputs@{ self, nixos-raspberrypi }: {
    nixosConfigurations.rpi5-simple-ssh = nixos-raspberrypi.lib.nixosSystem {
      modules = [ nixos-raspberrypi.nixosModules.raspberry-pi-5.base ./rpi5-simple-ssh.nix ];
      specialArgs = inputs;
    };
  };
}
```
and `rpi5-simple-ssh.nix` is:
```nix
{ config, pkgs, lib, ... }:
let singleUserName = "user";
in {
  networking.hostName = "rpi5-simple-ssh";
  services.openssh = {
    enable = true;
    settings.AllowUsers = [ singleUserName ];
    settings.PermitRootLogin = "no";
  };
  users = {
    mutableUsers = false;
    users.${singleUserName} = {
      isNormalUser = true;
      extraGroups = "wheel";
      openssh.authorizedKeys.keys = [
        "<my ssh key>"
      ];
    };
  };
}
```

This errors with:
```
nix-repl> :lf .
nix-repl> nixosConfigurations.rpi5-simple-ssh.config   
error:
       … while evaluating the attribute 'rpi5-simple-ssh.config'
         at /nix/store/qlyl20h0an2va5h9m8cavvgyhkvlhixa-source/lib/modules.nix:360:9:
          359|         options = checked options;
          360|         config = checked (removeAttrs config [ "_module" ]);
             |         ^
          361|         _module = checked (config._module);

       … while calling the 'seq' builtin
         at /nix/store/qlyl20h0an2va5h9m8cavvgyhkvlhixa-source/lib/modules.nix:360:18:
          359|         options = checked options;
          360|         config = checked (removeAttrs config [ "_module" ]);
             |                  ^
          361|         _module = checked (config._module);

       (stack trace truncated; use '--show-trace' to show the full trace)

       error: The option `boot.loader.raspberryPi' in module `/nix/store/qlyl20h0an2va5h9m8cavvgyhkvlhixa-source/nixos/modules/rename.nix' would be a parent of the following options, but its type `<no description>' does not support nested options.
        - option(s) with prefix `boot.loader.raspberryPi.bootPath' in module `/nix/store/ha19ys6xwln2qpwg7j6466ky27s2afh7-source/modules/system/boot/loader/raspberrypi'
        - option(s) with prefix `boot.loader.raspberryPi.bootPopulateCmd' in module `/nix/store/ha19ys6xwln2qpwg7j6466ky27s2afh7-source/modules/system/boot/loader/raspberrypi'
        - option(s) with prefix `boot.loader.raspberryPi.bootloader' in module `/nix/store/ha19ys6xwln2qpwg7j6466ky27s2afh7-source/modules/system/boot/loader/raspberrypi'
        - option(s) with prefix `boot.loader.raspberryPi.enable' in module `/nix/store/ha19ys6xwln2qpwg7j6466ky27s2afh7-source/modules/system/boot/loader/raspberrypi'
        - option(s) with prefix `boot.loader.raspberryPi.firmwarePackage' in module `/nix/store/ha19ys6xwln2qpwg7j6466ky27s2afh7-source/modules/system/boot/loader/raspberrypi'
        - option(s) with prefix `boot.loader.raspberryPi.firmwarePath' in module `/nix/store/ha19ys6xwln2qpwg7j6466ky27s2afh7-source/modules/system/boot/loader/raspberrypi'
        - option(s) with prefix `boot.loader.raspberryPi.firmwarePopulateCmd' in module `/nix/store/ha19ys6xwln2qpwg7j6466ky27s2afh7-source/modules/system/boot/loader/raspberrypi'
        - option(s) with prefix `boot.loader.raspberryPi.ubootPackage' in module `/nix/store/ha19ys6xwln2qpwg7j6466ky27s2afh7-source/modules/system/boot/loader/raspberrypi'
        - option(s) with prefix `boot.loader.raspberryPi.useGenerationDeviceTree' in module `/nix/store/ha19ys6xwln2qpwg7j6466ky27s2afh7-source/modules/system/boot/loader/raspberrypi'
        - option(s) with prefix `boot.loader.raspberryPi.variant' in module `/nix/store/ha19ys6xwln2qpwg7j6466ky27s2afh7-source/modules/system/boot/loader/raspberrypi'

```

schuelermine, 2025-06-11:
oh it’s probably overwriting the input that borks it

schuelermine, 2025-06-11:
yep

closed 2025-06-11 by schuelermine

rencire, 2025-09-18:
@schuelermine I have a similar issue testing the rpi zero 2 example.  How did you resolve this?

nvmd, 2025-09-19:
@rencire the simplest solution is don't overwrite `nixos-raspberrypi.inputs.nixpkgs.url = "github:NixOS/nixpkgs/nixos-24.11";`. Let `nixos-raspberrypi` use it own `nixpkgs`.

schuelermine, 2025-09-19:
Yeah that’s what worked