nvmd/nixos-raspberrypi #35: Device tree overlays are not applied

dankirkham, 2025-06-13:
I am using an rpi4 with uboot. I am trying to get a UART working on the header (GPIO14/15) so that the Pi can host a shield device with a TTL UART. I am specifically NOT trying to have a login prompt/tty available on the UART. I have tried a lot of things, but I *think* the issue is that device tree overlays are not actually applied.

To illustrate why I think this is happening, look what happens when I try to use the `disable-bt` overlay. (BT uses one of the UARTs, so disabling it frees it up for other usage)
```nix
  hardware.raspberry-pi.config.all.dt-overlays = {
    disable-bt = {
      enable = true;
      params = {};
    };
    "uart0" = {
      enable = true;
      params = {};
    };
  };
```

This propagates to the `/boot/firmware/config.txt` as expected.
```
dtoverlay=disable-bt

dtoverlay=
```

For reference, this is what the `disable-bt` overlay looks like:
```dts
/dts-v1/;

/ {
	compatible = "brcm,bcm2835";

	fragment@0 {
		target = <0xffffffff>;

		__overlay__ {
			status = "disabled";
		};
	};

	fragment@1 {
		target = <0xffffffff>;

		__overlay__ {
			pinctrl-names = "default";
			pinctrl-0 = <0xffffffff>;
			status = "okay";
		};
	};

	fragment@2 {
		target = <0xffffffff>;

		__overlay__ {
			status = "disabled";
		};
	};

	fragment@3 {
		target = <0xffffffff>;

		__overlay__ {
			brcm,pins;
			brcm,function;
			brcm,pull;
		};
	};

	fragment@4 {
		target = <0xffffffff>;

		__overlay__ {
			brcm,pins;
			brcm,function;
			brcm,pull;
		};
	};

	fragment@5 {
		target-path = "/aliases";

		__overlay__ {
			serial0 = "/soc/serial@7e201000";
			serial1 = "/soc/serial@7e215040";
		};
	};

	__fixups__ {
		uart1 = "/fragment@0:target:0";
		uart0 = "/fragment@1:target:0";
		uart0_pins = "/fragment@1/__overlay__:pinctrl-0:0", "/fragment@3:target:0";
		bt = "/fragment@2:target:0";
		bt_pins = "/fragment@4:target:0";
	};
};
```

I am still pretty new to device trees, but I believe `fragment@2` is what should be disabling bluetooth.
```dts
fragment@2 {
	target = <0xffffffff>;

	__overlay__ {
		status = "disabled";
	};
};
```

However, when I inspect `/sys/firmware/fdt` I do not see the bluetooth status is still `okay`:
```dts
serial@7e201000 {
        compatible = "arm,pl011", "arm,primecell";
        reg = <0x7e201000 0x200>;
        interrupts = <0x00 0x79 0x04>;
        clocks = <0x08 0x13 0x08 0x14>;
        clock-names = "uartclk", "apb_pclk";
        arm,primecell-periphid = <0x341011>;
        status = "okay";
        cts-event-workaround;
        pinctrl-names = "default";
        pinctrl-0 = <0x09 0x0a>;
        uart-has-rtscts;
        skip-init;
        phandle = <0x43>;

        bluetooth {
                compatible = "brcm,bcm43438-bt";
                max-speed = <0x2dc6c0>;
                shutdown-gpios = <0x0b 0x00 0x00>;
                local-bd-address = [5d 14 12 dd 3a d8];
                fallback-bd-address;
                status = "okay";
                phandle = <0x41>;
        };
};
```

Unless there is some nuance to how I am using `dtc` or `/sys/firmware/fdt` isn't what I think it is, overlays are not being applied.

nvmd, 2025-06-18:
Hm, could `uart0` overlay conflict with `disable-bt`?

I use `disable-bt` on my own RPi5, and I do get (with `nix run nixpkgs#dtc -- -I fs -O dts /proc/device-tree`) the correct status, i.e. the overlay did apply.
```
bluetooth {
   ...
   status = "disabled";
   ...
}
```

dankirkham, 2025-06-19:
Thanks for the response!

I didn't mean to include the `uart0` setting in my example, as it is superfluous because I tried it with and without and it still did not work.

I tried your method of reading the device tree and got the same result, so that is not this issue.

As for kernelboot, in order to enable it, I had to allow overriding of another setting as seen in #38.

Once I enabled it I encountered another issue, in addition to overlays still not working. It seems to still use `uboot` in some fashion, even with `kernel=kernel.img` set for `kernelboot`. In my `/boot/firmware/cmdline.txt` it correctly had `kernelboot` stuff, but upon reboot `/proc/cmdline` had the old `uboot` stuff. I decided to rename the `extlinux` folder and the `uboot*.bin` and now it doesn't boot at all!

dankirkham, 2025-06-19:
My `/boot/firmware` partition was not actually being mounted. 🤦

Both `kernelboot` and `uboot` options have correct overlays now!