nvmd/nixos-raspberrypi #41: Unable to use `nixos-raspberrypi.lib.nixosSystemFull`

MatthiasvB, 2025-06-26:
I'm trying to migrate my homelab from an old x86 laptop to my rpi5. For now, I've dummed it down a bit, which allows building with `nixos-raspberrypi.lib.nixosSystem`, but some things crash at runtime (like postgres, which is very important). I had some hopes of fixing some of these by using the overlays defined in `nixos-raspberrypi.lib.nixosSystemFull`, but when switching to that I get

```
error:
       … while calling the 'head' builtin
         at /nix/store/mvnfzv7y69lbr56vdhrhdqkv5b7qhsr2-source/lib/attrsets.nix:1534:13:
         1533|           if length values == 1 || pred here (elemAt values 1) (head values) then
         1534|             head values
             |             ^
         1535|           else

       … while evaluating the attribute 'value'
         at /nix/store/mvnfzv7y69lbr56vdhrhdqkv5b7qhsr2-source/lib/modules.nix:1084:7:
         1083|     // {
         1084|       value = addErrorContext "while evaluating the option `${showOption loc}':" value;
             |       ^
         1085|       inherit (res.defsFinal') highestPrio;

       … while evaluating the option `system.build.toplevel':

       … while evaluating definitions from `/nix/store/mvnfzv7y69lbr56vdhrhdqkv5b7qhsr2-source/nixos/modules/system/activation/top-level.nix':

       … while evaluating the option `warnings':

       … while evaluating definitions from `/nix/store/mvnfzv7y69lbr56vdhrhdqkv5b7qhsr2-source/nixos/modules/system/boot/systemd.nix':

       … while evaluating the option `systemd.services.immich-server.serviceConfig':

       … while evaluating definitions from `/nix/store/mvnfzv7y69lbr56vdhrhdqkv5b7qhsr2-source/nixos/modules/services/web-apps/immich.nix':

       (stack trace truncated; use '--show-trace' to show the full, detailed trace)

       error: function 'anonymous lambda' called with unexpected argument 'version'
       at /nix/store/ha19ys6xwln2qpwg7j6466ky27s2afh7-source/pkgs/ffmpeg_7-rpi.nix:1:1:
            1| { lib, fetchFromGitHub
             | ^
            2| , callPackage
```

I can't say I quite understand the issue. Perhaps this would be fixed by just adding the `, ...` to the attr set argument in `ffmpeg_7-rpi.nix`, but what is causing the extra argument to be passed I can only wildly speculate. That speculation is that in 25.05 the arguments were changed and this repo hasn't been adjusted to work with the new version, yet?

nvmd, 2025-06-27:
Hi!

> but some things crash at runtime (like postgres, which is very important).

Could that be caused by some binary data, non-portable across different architectures x86->arm. I presume that could be the case, especially for Postgres. Did you perform some migration procedure?

Now, to the `nixos-raspberrypi` related stuff:

> I'm trying to migrate my homelab from an old x86 laptop to my rpi5. For now, I've dummed it down a bit, which allows building with `nixos-raspberrypi.lib.nixosSystem`, but some things crash at [...]

Using `nixosSystem` is actually more safe than `nixosSystemFull` because the latter applies overlays with raspberry-optimized packages everywhere transitively and unconditionally, so to say. That means, on one hand, possible benefits through the system, on the hand, it can break something and this isn't something we can thoroughly test now.

The error you're getting is (almost) an example of that, actually. You're using `immich` which pulls `jellyfin-ffmpeg`, which then uses its own source with the `ffmpeg_7` derivation (see here https://github.com/NixOS/nixpkgs/blob/nixos-unstable/pkgs/by-name/je/jellyfin-ffmpeg/package.nix#L11). Combining ffmpeg raspberry's patches with jellyfin's patches may work great, or may break something. You'll need to test that.

> `...` to the attr set argument in `ffmpeg_7-rpi.nix`, but what is causing the extra argument to be passed I can only wildly speculate.

I think it should work at least in this particular case. I'll get back to it later, we'll probably need at least a little bit better solution in the long run.

MatthiasvB, 2025-07-18:
So, I kept getting stuck pretty early on in every attempt I made to make this think work. The biggest issue I seem to be having is dealing with jemalloc vs 16k page size. I am just not able to make this work. But, it should be possible to switch the kernel to 4k page size. Is this possible somehow with the work in this repo? Or is there generally a way to achieve this?

Also, this repo still seems to suggest that uboot still isn't usable with the Pi 5, yet I have a friend with a Pi 5 that _is_ runnig uboot. But he's using the `raspberry-pi-nix` repo, which is archived. Is there support for this here?

nvmd, 2025-07-20:
> So, I kept getting stuck pretty early on in every attempt I made to make this think work. The biggest issue I seem to be having is dealing with jemalloc vs 16k page size. I am just not able to make [...]

`rpi4` kernel has 4k page size, you can use that. See this as an example: https://github.com/nvmd/nixos-raspberrypi-demo/blob/a9031b6d2d68d0fc43c39b85905b0e924f60dd7e/flake.nix#L282.
I think that is the recommended way to have 4k pages on rpi5.

Diving a little deeper, `rpi4` and `rpi5` kernels use different configuration options:
- `rpi4` uses [`bcm2711_defconfig`](https://raw.githubusercontent.com/raspberrypi/linux/refs/tags/stable_20240529/arch/arm64/configs/bcm2711_defconfig)
- `rpi5` uses [`bcm2712_defconfig`](https://raw.githubusercontent.com/raspberrypi/linux/refs/tags/stable_20240529/arch/arm64/configs/bcm2712_defconfig)

Meanwhile, 16k vs 4k page size is actually controlled by a single `CONFIG_ARM64_16K_PAGES=y` flag.
Should you need something specific to `bcm2712` on your system, you could probably toggle the flag and recompile the kernel, but I don't know if this will result in a sound configuration.

> Also, this repo still seems to suggest that uboot still isn't usable with the Pi 5, yet I have a friend with a Pi 5 that _is_ runnig uboot. But he's using the `raspberry-pi-nix` repo, which is archived. Is there support for this here?

Yes, there is.
Last time I checked uboot had _limited_ support for Pi 5, but nonetheless usable. That depends on the usecase, really. I just believe that "kernelboot" is a safer default choice for RPi5 for now.
Regardless, one can use `uboot` by setting `boot.loader.raspberryPi.bootloader = "uboot"`

pythonpoet, 2025-10-05:
Hi is there any progress on that? I'm having the same issue