nvmd/nixos-raspberrypi #44: Pi 5 active cooler/pwm control not working

reinux, 2025-07-02:
I can't be 100% sure this is anything to do with this project, but just in case...

I'm not seeing `cooling_device0`, under `/sys/class/thermal`, which suggests to me it's something to do wtih the firmware/kernel.

There's also nothing under `/sys/class/pwm`

However, I'm able to manually turn the fan on/off using `pinctrl FAN_PWM op dl`.

Any help would be appreciated.

Thanks!

reinux, 2025-07-03:
Hey thanks for the reply. Here's what uname tells me:

```uname
rei@onigiri ~> uname -a
Linux onigiri 6.12.25 #1-NixOS SMP PREEMPT Tue Jan  1 00:00:00 UTC 1980 aarch64 GNU/Linux
```

I think the rest of the config should also be quite standard. Just to be sure, I set the raspberrypi boot loader explicitly (I'm assuming it uses kernelboot?), but it didn't seem to make a difference:

```nix
  system.nixos.tags = [
    "raspberry-pi-${config.boot.loader.raspberryPi}"
    config.boot.loader.raspberryPi.bootloader
    config.boot.kernelPackages.kernel.version
  ];
```

Hardware wise, aside from the fact that I'm using a Geeekpi nvme+PoE hat (the fan came with the hat), it's also quite standard.

nvmd, 2025-07-03:
> I set the raspberrypi boot loader explicitly (I'm assuming it uses kernelboot?), but it didn't seem to make a difference:

So what is the value of `config.boot.loader.raspberryPi.bootloader` exactly (you can check it using `nix repl`)? `system.nixos.tags` just adds the tags to the name of the system configuration.

> "raspberry-pi-${config.boot.loader.raspberryPi}"

This line looks like it shouldn't evaluate, because it shouldn't be able to convert `attrset` to `string`. The correct value should be `config.boot.loader.raspberryPi.variant`.
Is it the excerpt from the configuration you're actually running?

You could also try whether you'll get the fan controls working for you with linux 6.6.31 – it's available in the binary cache. See: https://github.com/nvmd/nixos-raspberrypi-demo/blob/2847963e7555fc412c1d0f37bb48c761e78f350d/flake.nix#L283

reinux, 2025-07-04:
I honestly couldn't tell you where I got that code snippet from, because you're right, that reads `"raspberry-pi-${cfg.variant}"` in my config lol.

`config.boot.loader.raspberryPi.variant` is `kernelboot`.

I have it running 6.6.31 right now, which created an entry `/sys/class/pwm/pwmchip0`, but no `cooling_device` under `/sys/class/thermal`. Fan's still going at 100%.

Another thing I just tried was to remove the hardware-configuration.nix import, which I needed at one point for reasons that I can't recall. (I think it wanted `nixpkgs.hostPlatform` to be defined.)

Does the fan go when you use the installer? Because I didn't change anything except my public key when I built the installer, and the fan was going the whole time as that was running too.

closed 2025-07-04 by reinux

reinux, 2025-07-04:
> You can actually update board's EEPROM (firmware) with rpi-eeprom-update from pkgs.raspberrypi-eeprom.

Good to know. Thanks!