nvmd/nixos-raspberrypi #48: camera packages not working

adminy, 2025-07-07:
```
[0:00:49.764240350] [1054]  INFO IPAManager ipa_manager.cpp:137 libcamera is not installed. Adding '/nix/store/src/ipa' to the IPA search path
[0:00:49.769436425] [1054]  INFO Camera camera_manager.cpp:326 libcamera v0.5.0
[0:00:49.780845180] [1059]  INFO RPI pisp.cpp:720 libpisp version 1.2.1
terminate called after throwing an instance of 'std::runtime_error'
  what():  BE: Could not find config json file: /nix/store/n3xa0k7fjxx3r52gs8qiz5l7v9g0kgr9-libpisp-1.2.1/lib/libpisp/backend/backend_default_config.json

```
Even though I have libcamera installed and libpisp is there.

adminy, 2025-07-20:
Also i2c interfaces aren't loaded when you run `lsmod`

nvmd, 2025-07-20:
> Also i2c interfaces aren't loaded when you run `lsmod`

How did you configure it? This should be a separate issue, though. Should you still have this i2c problem, please create a new issue describing your situation in more detail.

nvmd, 2025-07-20:
I'll close this issue, as it's unrelated to `nixos-raspberrypi` itself, it should either be escalated to
- [libpisp](https://github.com/raspberrypi/libpisp/), because their method of determining the path to the configuration file doesn't work on nixos. See https://github.com/raspberrypi/libpisp/blob/2fe908435d3f4890f422996b364ad0becec65867/src/libpisp/backend/backend_default_config.cpp#L322
- or `nixpkgs`, because this flake uses `libpisp` derivation from there, and the derivation doesn't accommodate for the specifics of `libpisp`.

Namely, `libpisp` relies on meson stripping `RPATH`s in binaries, which doesn't happen on NixOS (see https://github.com/NixOS/nixpkgs/pull/35513 for more details). Thus, `libpisp` sets the wrong path to its default configuration file.

closed 2025-07-20 by nvmd

adminy, 2025-07-20:
Thank you for looking into this. It's best that we address this issue upstream in the nixpkgs then. There must be a meson example on how the nixpkgs folks address this kind of problem, I doubt libpisp is the first one to encounter this particular issue. Linux packages are full of "assumptions" all over the place. I'll check if there is a standard solution to this. 

As for the i2c issue, I noticed some modules are missing even though I've got i2c enabled in the pi config. Also the /dev/i2c* is non existent. I'll investigate this further before I open a new issue.

adminy, 2025-07-26:
I've finally been able to figure out this meson build mess and get libpisp built without runpath. #55 updated. I gotta say that its not great on how many things get rebuilt because of this one library. 

Thanks @nvmd for `hardware.i2c.enable`, I would have never figured it out. Now the camera device is working!

adminy, 2025-07-29:
@nvmd I have opened a [PR#429288](https://github.com/NixOS/nixpkgs/pull/429288) upstream in nixpkgs.