nvmd/nixos-raspberrypi #49: CM5 stuck at NixOS Stage 1 with encrypted root

graureiher, 2025-07-08:
Thanks a lot for your work on this project! I am currently trying to get NixOS running on a CM5 with an encrypted root partition. Unfortunately, the boot process is currently stuck at Stage 1 (this is basically the same problem as in #14):

```
Starting systemd-udevd version 257.3
```

Here is my config:

`flake.nix`:
```
{
  inputs = {
    nixpkgs.url = "github:nixos/nixpkgs/nixos-unstable";
    systems.url = "github:nix-systems/default-linux";

    hardware.url = "github:nixos/nixos-hardware";
    nixos-raspberrypi.url = "github:nvmd/nixos-raspberrypi/main";
  };

  outputs = {
    self,
    nixpkgs,
    systems,
    nixos-raspberrypi,
    ...
  } @ inputs: let
    inherit (self) outputs;
    lib = nixpkgs.lib;
    forEachSystem = f: lib.genAttrs (import systems) (system: f pkgsFor.${system});
    pkgsFor = lib.genAttrs (import systems) (
      system:
        import nixpkgs {
          inherit system;
          config.allowUnfree = true;
        }
    );
  in {
    inherit lib;

    nixosConfigurations = {
      # CM5 Testing
      krameri = nixos-raspberrypi.lib.nixosSystem {
	modules = [./hosts/krameri];
	specialArgs = {
	 inherit inputs outputs nixos-raspberrypi;
	};
      };
    };
  };

  nixConfig = {
    extra-substituters = [
      "https://nixos-raspberrypi.cachix.org"
    ];
    extra-trusted-public-keys = [
      "nixos-raspberrypi.cachix.org-1:4iMO9LXa8BqhU+Rpg6LQKiGa2lsNh/j2oiYLNOQ5sPI="
    ];
  };
}
```

And here are the corresponding default.nix and hardware-configuration.nix files for the cm5 host:

`default.nix`
```
{
  inputs,
  lib,
  ...
}: {
  imports = [
    inputs.nixos-raspberrypi.nixosModules.raspberry-pi-5.base
    inputs.nixos-raspberrypi.nixosModules.raspberry-pi-5.display-vc4

    ./hardware-configuration.nix
  ];

  networking = {
    hostName = "krameri";
    useDHCP = true;
  };

  boot.initrd = {
    kernelModules = [ "mmc_block" "uas" "usbcore" "usb_storage" "vfat" "nls_cp437" "nl_iso8859_1" ];
    supportedFilesystems = [ "btrfs" "vfat" ];
  };

  time.timeZone = "Europe/Berlin";
  users.users.root = {
    openssh.authorizedKeys.keys = [
      ...
    ];
  };

  networking.firewall.allowedTCPPorts = [ 22 ];
  services.openssh.enable = true;

  system.stateVersion = "25.05";
}
```

`hardware-configuration.nix`
```
{
  config,
  lib,
  pkgs,
  modulesPath,
  ...
}: {
fileSystems."/" =
    { device = "/dev/disk/by-uuid/8da53a89-41e8-4996-84f1-dbad195f1116";
      fsType = "btrfs";
      options = [ "subvol=root" ];
    };

  boot.initrd.luks.devices."crypted".device = "/dev/disk/by-uuid/d4e096a1-f41f-4bbc-bcd8-1755e9e65c10";

  fileSystems."/boot/firmware" =
    { device = "/dev/disk/by-uuid/A775-6574";
      fsType = "vfat";
      options = [ "fmask=0022" "dmask=0022" ];
    };

  fileSystems."/nix" =
    { device = "/dev/disk/by-uuid/8da53a89-41e8-4996-84f1-dbad195f1116";
      fsType = "btrfs";
      options = [ "subvol=nix" ];
    };

  fileSystems."/persist" =
    { device = "/dev/disk/by-uuid/8da53a89-41e8-4996-84f1-dbad195f1116";
      fsType = "btrfs";
      options = [ "subvol=persist" ];
    };

  fileSystems."/swap" =
    { device = "/dev/disk/by-uuid/8da53a89-41e8-4996-84f1-dbad195f1116";
      fsType = "btrfs";
      options = [ "subvol=swap" ];
    };

  swapDevices = [ ];
}
```

Do you see any issues with that configuration that would lead to the system being stuck? Or is there any other problem, maybe with using a CM5? Thanks in advance!

graureiher, 2025-07-08:
I can provide you the boot log I obtained via serial, I hope this is helpful:
```
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x414fd0b1]
[    0.000000] Linux version 6.12.25 (nixbld@localhost) (gcc (GCC) 14.2.1 20250322, GNU ld (GNU Binutils) 2.44) #1-NixOS SMP PREEMPT Tue Jan  1 00:00:00 UTC 1980
[    0.000000] KASLR enabled
[    0.000000] random: crng init done
[    0.000000] Machine model: Raspberry Pi Compute Module 5 Rev 1.0
[    0.000000] efi: UEFI not found.
[    0.000000] Reserved memory: created CMA memory pool at 0x000000003a000000, size 64 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] OF: reserved mem: 0x000000003a000000..0x000000003dffffff (65536 KiB) map reusable linux,cma
[    0.000000] OF: reserved mem: 0x0000000000000000..0x000000000007ffff (512 KiB) nomap non-reusable atf@0
[    0.000000] OF: reserved mem: 0x000000003fd16320..0x000000003fd16389 (0 KiB) nomap non-reusable nvram@0
[    0.000000] NUMA: Faking a node at [mem 0x0000000000000000-0x00000001ffffffff]
[    0.000000] Faking node 0 at [mem 0x0000000000000000-0x000000003fffffff] (1024MB)
[    0.000000] Faking node 1 at [mem 0x0000000040000000-0x000000007fffffff] (1024MB)
[    0.000000] Faking node 2 at [mem 0x0000000080000000-0x00000000bfffffff] (1024MB)
[    0.000000] Faking node 3 at [mem 0x00000000c0000000-0x00000000ffffffff] (1024MB)
[    0.000000] Faking node 4 at [mem 0x0000000100000000-0x000000013fffffff] (1024MB)
[    0.000000] Faking node 5 at [mem 0x0000000140000000-0x000000017fffffff] (1024MB)
[    0.000000] Faking node 6 at [mem 0x0000000180000000-0x00000001bfffffff] (1024MB)
[    0.000000] Faking node 7 at [mem 0x00000001c0000000-0x00000001ffffffff] (1024MB)
[    0.000000] NODE_DATA(0) allocated [mem 0x3fbfc780-0x3fbfffff]
[    0.000000] NODE_DATA(1) allocated [mem 0x7fffc780-0x7fffffff]
[    0.000000] NODE_DATA(2) allocated [mem 0xbfffc780-0xbfffffff]
[    0.000000] NODE_DATA(3) allocated [mem 0xffffc780-0xffffffff]
[    0.000000] NODE_DATA(4) allocated [mem 0x13fffc780-0x13fffffff]
[    0.000000] NODE_DATA(5) allocated [mem 0x17fffc780-0x17fffffff]
[    0.000000] NODE_DATA(6) allocated [mem 0x1bfffc780-0x1bfffffff]
[    0.000000] NODE_DATA(7) allocated [mem 0x1ff398780-0x1ff39bfff]
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x00000000ffffffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   [mem 0x0000000100000000-0x00000001ffffffff]
[    0.000000]   Device   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000000007ffff]
[    0.000000]   node   0: [mem 0x0000000000080000-0x000000003fbfffff]
[    0.000000]   node   1: [mem 0x0000000040000000-0x000000007fffffff]
[    0.000000]   node   2: [mem 0x0000000080000000-0x00000000bfffffff]
[    0.000000]   node   3: [mem 0x00000000c0000000-0x00000000ffffffff]
[    0.000000]   node   4: [mem 0x0000000100000000-0x000000013fffffff]
[    0.000000]   node   5: [mem 0x0000000140000000-0x000000017fffffff]
[    0.000000]   node   6: [mem 0x0000000180000000-0x00000001bfffffff]
[    0.000000]   node   7: [mem 0x00000001c0000000-0x00000001ffffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x000000003fbfffff]
[    0.000000] Initmem setup node 1 [mem 0x0000000040000000-0x000000007fffffff]
[    0.000000] Initmem setup node 2 [mem 0x0000000080000000-0x00000000bfffffff]
[    0.000000] Initmem setup node 3 [mem 0x00000000c0000000-0x00000000ffffffff]
[    0.000000] Initmem setup node 4 [mem 0x0000000100000000-0x000000013fffffff]
[    0.000000] Initmem setup node 5 [mem 0x0000000140000000-0x000000017fffffff]
[    0.000000] Initmem setup node 6 [mem 0x0000000180000000-0x00000001bfffffff]
[    0.000000] Initmem setup node 7 [mem 0x00000001c0000000-0x00000001ffffffff]
[    0.000000] On node 1, zone DMA: 256 pages in unavailable ranges
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv1.1 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: MIGRATE_INFO_TYPE not supported.
[    0.000000] psci: SMC Calling Convention v1.2
[    0.000000] percpu: Embedded 27 pages/cpu s259864 r65536 d116968 u442368
[    0.000000] Detected PIPT I-cache on CPU0
[    0.000000] CPU features: detected: Virtualization Host Extensions
[    0.000000] CPU features: detected: Spectre-v4
[    0.000000] CPU features: detected: Spectre-BHB
[    0.000000] CPU features: kernel page table isolation forced ON by KASLR
[    0.000000] CPU features: detected: Kernel page table isolation (KPTI)
[    0.000000] CPU features: detected: SSBS not fully self-synchronizing
[    0.000000] alternatives: applying boot alternatives
[    0.000000] Kernel command line: reboot=w coherent_pool=1M 8250.nr_uarts=1 pci=pcie_bus_safe cgroup_disable=memory numa_policy=interleave  numa=fake=8 system_heap.max_order=0 smsc95xx.macaddr=2C:CF:67:FC:C9:3F vc_mem.mem_base=0x3fc00000 vc_mem.mem_size=0x40000000  console=ttyAMA0,115200 console=tty1 loglevel=7 lsm=landlock,yama,bpf init=/nix/store/17pk8pbl5giid4zpz61qcbszvv584d9d-nixos-system-krameri-25.05.20250704.df3e335/init
[... 363 lines cut ...]
[    4.126771]    8regs           : 17328 MB/sec
[    4.131323]    32regs          : 17407 MB/sec
[    4.135822]    arm64_neon      : 22603 MB/sec
[    4.140177] xor: using function: arm64_neon (22603 MB/sec)
[... cut ...]
[    4.424704] vc4-drm axi:gpu: Adding to iommu group 1
[    4.429662] vc4-drm axi:gpu: bcm2712_iommu_attach_dev: MMU 1000005200.iommu
[    4.438977] vc4-drm axi:gpu: bound 107c580000.hvs (ops vc4_hvs_ops [vc4])
[   14.565741] vc4-drm axi:gpu: bound 107c580000.hvs (ops vc4_hvs_ops [vc4])
[   14.574360] platform axi:gpu: deferred probe pending: (reason unknown)
```

Let me know, if you need any additional logs.

I want the system to ask me for the encryption password, unattended boot is not needed on this system.

nvmd, 2025-07-20:
@graureiher when you still have a little time, I just want to clarify, whether you've been using the default `kernelboot` bootloader or setting `uboot` explicitly somewhere in your configuration?

graureiher, 2025-07-20:
> I'll keep this issue open, because this definitely needs an additional research. There are scenarios when disabling `vc4-kms-v3d` isn't really an option. I wonder if it's actually an upstream problem or not, and how can we work around that.

Let me know if I can help debugging the issue. I have a spare CM5 left to tinker on.

> [@graureiher](https://github.com/graureiher) when you still have a little time, I just want to clarify, whether you've been using the default `kernelboot` bootloader or setting `uboot` explicitly somewhere in your configuration?

I am using the default `kernelboot` bootloader right now. I also tried `uboot`, but this did not have any noticeable effect on the issue at hand.

pwaller, 2025-07-20:
I'm also struggling to get my rpi5 to boot. It also gets to a similar position:
```
[    3.383889] Run /init as init process
```

I'm using: `boot.loader.raspberryPi.enable = true;``

nvmd, 2025-07-20:
Hi @pwaller!
Based on the info we have so far, it looks like it's an unrelated issue, I'll be happy to assist you in a newly created one, duplicating the content of your original comment :)

That would be nice if you could also answer some of the questions below in the issue description:

* Did you try booting the unchanged `rpi5-installer` image provided by the flake?
* Does it boot without `rpi-digiampplus` overlay?
* Has this particular nixos system ever booted on RPi5? Maybe there're some changes that you've made that could have broken it?
* May you could provide more complete boot log?

I see the `config.txt` differs from the default one, and also appears to be manually edited: it should have the following lines in this particular order:
```
dtoverlay=rpi-digiampplus
dtoverlay=
```

Otherwise, to factor out the broken `config.txt` file generator, please provide `:p nixosConfigurations.<your rpi5 system>.config.hardware.raspberry-pi.config-generated` and `:p nixosConfigurations.<your rpi5 system>.config.hardware.raspberry-pi.config` from `nix repl`.

graureiher, 2025-08-28:
I've been revisiting this issue while working on getting NixOS running on a uConsole from ClockworkPi. The device requires the `vc3-kms-v3d-pi5` overlay for the display to function properly.

In the related issue opened by @pwaller, he made an observation that might shed some light on the root cause (https://github.com/nvmd/nixos-raspberrypi/issues/57#issuecomment-3167866481):

> One more observation. With dtoverlay=vc4-kms-v3d, if there is a problem with the second stage (for example, failure to mount the root partition due to misconfiguration or typo or the boot order issue I had above), it manifests as a hang somewhere around these messages

My suspicion is that something similar is happening in my case. Specifically, the password prompt seems to halt the boot process, just like it does when the boot order is misconfigured. I'm by no means an expert on the Linux or NixOS boot process, but it might be worth investigating why system messages (like a boot order issue or password prompt) aren't displayed when the vc3 overlay is active.

pwaller, 2025-08-28:
I presume what's happening is that the system is trying to transition to graphical mode, but something in the act of transitioning is failing (because e.g. some critical files are effectively missing). I guess there isn't much that could be done in that scenario other than to have a separate serial console to get the messages out. If the causes of the failure were understood, perhaps they could be defended against, but presumably there could be any number of possible causes.

miangraham, 2025-10-04:
Here's my first confirmed working fix:

```nix
boot.blacklistedKernelModules = [ "vc4" ];
systemd.services.modprobe-vc4 = {
  serviceConfig = {
    Type = "oneshot";
    User = "root";
  };
  before = [ "multi-user.target" ];
  wantedBy = [ "multi-user.target" ];
  script = "/run/current-system/sw/bin/modprobe vc4";
};
```

This delays the initialization of vc4 until after the root volume is decrypted. The result is that the console retains its initial display mode during LUKS password entry, then switches modes properly later on, auto-detecting displays and producing the expected resolutions for graphical environments.

Blacklisting the module to negate the entry in `boot.initrd.availableKernelModules` also prevents it from autoloading even if added to `boot.kernelModules`, so a modprobe command is required, resulting in the one-line service script. It's not pretty but it works.

A nice-to-have for the project would be an option in `modules/raspberrypi` to move vc4 from `boot.initrd.availableKernelModules` to `boot.kernelModules`, fixing the problem while still letting it autoload. This would obviate my modprobe service addition, letting people resolve the issue with a single config line.

slcn-dxd, 2025-10-04:
Great news and thanks a lot for developing the fix @miangraham! Maybe the `vc4` module could be moved to `boot.kernelModules` altogether? At first glance, I don't see a reason why it should be needed in the first boot stage anyway. What do you think, @nvmd?

slcn-dxd, 2025-10-05:
That's true, maybe the module could be removed from `availableKernelModules` conditionally like
```
let
  luksConfigured = config.boot.initrd.luks.devices != {};
in {
  boot = {
    initrd.availableKernelModules = [
      "xhci_pci"
      "usbhid"
      "usb_storage"
      "pcie_brcmstb"
      "reset-raspberrypi"
    ] ++ lib.optional (!luksConfigured) "vc4";

    kernelModules = lib.optional luksConfigured "vc4";
  };
```

elmarx, 2026-11-03:
for me (CM5, ZFS encrypted root), adding `i2c_brcmstb` to initrd did the trick, because vc4 kicks out the firmware framebuffer, but needs DDC I2C, which is only available once `i2c_brcmstb` hase been loaded… which is only available in stage 2.

I.e. test with `boot.initrd.availableKernelModules = [ "i2c_brcmstb" ];`

closed 2026-11-15 by nvmd