nvmd/nixos-raspberrypi #51: Extending `lib` does not work on raspberry pi even tho it works on regular x86-64 nix config

ItsLiyua, 2025-07-12:
I'm trying to extend the `lib` attrset with some custom functions.
The code I wrote works fine for my x86 devices but when trying to build raspberry pi configs it doesn't recognize my custom functionality.

This is the relevant code in the `flake.nix` file:
```nix
=
    {
      nixpkgs,
      nixos-raspberrypi,
      ...
    }@inputs:
    let
      inputConfigs = with inputs; [ ... ];
      inputOverlays = with inputs; [ ... ];
      mkSysConfig =
        mainRepo: cfg:
        let
          lib = mainRepo.lib.extend (self: super: { liyua = import ./lib { inherit (nixpkgs) lib; }; });
        in
        lib.nixosSystem {
          specialArgs = inputs;
          modules = inputConfigs ++ [
            { nixpkgs.overlays = inputOverlays; }
            ...
            cfg
          ];
        };
    in
    {
      nixosConfigurations = {
        t480 = mkSysConfig nixpkgs ./hosts/t480;
        rpi5-1 = mkSysConfig nixos-raspberrypi ./hosts/pi/rpi5-1;
        ...
      };
    };
}
```

The file `lib/default.nix` contains this:
```nix
{ lib, ... }:
{
  relativeToRoot = lib.path.append ../.;
}
```

ajaxbits, 2025-08-15:
Same boat here

ItsLiyua, 2025-08-15:
For the time I fixed it by not extending lib and instead adding an extra parameter called `myLib` with my options to `specialArgs`. Still tho. Extending `lib` would be more elegant.