nvmd/nixos-raspberrypi #54: Build takes forever and then crashes during tests, isn't using cache

reinux, 2025-07-19:
Sorry to keep opening more issues. I'm trying nixosSystemFull to see if it addresses [the GSM issue](https://github.com/nvmd/nixos-raspberrypi/issues/53), but it takes hours to build and then crashes at random points due to tests timing out. In my last attempt, it crashed here:

```
[4/481/535 built, 1421 copied (9109.6/9114.7 MiB), 2461.0 MiB DL] building gjs-1.84.2 (configurePhase): Program /nix/store/rerror: builder for '/nix/store/68ydhzgqby3496a72hss4lv3c0j4vjpc-gjs-1.84.2.drv' failed with exit code 6;
       last 25 log lines:
       > 72/78 gjs:Debugger / print command                     OK             15.55s   1 subtests passed
       > 73/78 gjs:Debugger / until command                     OK             11.95s   1 subtests passed
       > 74/78 gjs:Debugger / sourcemap separate                OK             10.82s   1 subtests passed
       > 75/78 gjs:Debugger / sourcemap inlined module          OK              8.72s   1 subtests passed
       > 76/78 gjs:Debugger / sourcemap separate module         OK             12.39s   1 subtests passed
       > 77/78 gjs:Debugger / throw ignored                     OK              9.36s   1 subtests passed
       > 78/78 gjs:Debugger / sourcemap inlined                 OK             11.84s   1 subtests passed
       >
       > Summary of Failures:
       >
       > 21/78 gjs:Scripts / CommandLine                TIMEOUT        31.16s   killed by signal 15 SIGTERM
       > 22/78 gjs:C / Internal API tests               TIMEOUT        31.48s   killed by signal 15 SIGTERM
       > 23/78 gjs:JS / GIMarshalling                   TIMEOUT        31.13s   killed by signal 15 SIGTERM
       > 40/78 gjs:JS / Regress                         TIMEOUT        30.30s   killed by signal 15 SIGTERM
       > 49/78 gjs:C / API tests                        TIMEOUT        61.26s   killed by signal 15 SIGTERM
       > 51/78 gjs:JS / GObjectDestructionAccess        TIMEOUT        30.24s   killed by signal 15 SIGTERM
       >
       > Ok:                 72
       > Expected Fail:      0
       > Fail:               0
       > Unexpected Pass:    0
       > Skipped:            0
       > Timeout:            6
       >
       > Full log written to /build/gjs-1.84.2/build/meson-logs/testlog.txt
       For full logs, run:
         nix log /nix/store/68ydhzgqby3496a72hss4lv3c0j4vjpc-gjs-1.84.2.drv
```

Cachix is set up in my build inputs. I'm not using any overlays, so I'm not too sure why it's not using the cache at all.

adminy, 2025-07-19:
For me its running out of RAM.

zacharyweiss, 2025-09-10:
I've been running into the same with the vanilla `nixos-raspberrypi.lib.nixosSystem` on both `main` and `develop`; cache misses (even on the kernel, despite no overlays / modifications touching) and the gjs check timeouts. On the latter, resources not the issue for me; building on an overpowered x86 machine through QEMU (`boot.binfmt.emulatedSystems = ["aarch64-linux"];` in the build machine's config).

A workaround for others; one can apply a gjs overlay disabling the tests & checks:
```nix
  nixpkgs.overlays = [
    (_: prev: {
      gjs = (prev.gjs.override {installTests = false;}).overrideAttrs {doCheck = false;};
    })
  ];
```
Alternatively, I believe there'd be a way to disable/extend the timeout through an overlay on the checkPhase; didn't investigate deeply.