nvmd/nixos-raspberrypi #57: Raspberry Pi 5 boot hang

pwaller, 2025-07-20:
Forking from https://github.com/nvmd/nixos-raspberrypi/issues/49#issuecomment-3094809228.

I'm also struggling to get my rpi5 to boot. It also gets to a similar position, shortly after:

```
[    3.383889] Run /init as init process
```

I'm trying to boot off the sd card at the moment. The last message is something like `vc4-drm` ... `axi:gpu` ...`bcm2712_iommu_of_xlate: MMU 1000005200.iommu`.

I'm using: `boot.loader.raspberryPi.enable = true;`, and have't knowingly configured something outside of the default boot choices.

Here's my current config.txt if it helps:
```
[all]
arm_64bit=1
arm_boost=1
avoid_warnings=1
camera_auto_detect=1
disable_fw_kms_setup=1
disable_overscan=1
display_auto_detect=1
enable_uart=1
kernel=kernel.img
max_framebuffers=2
dtoverlay=
dtoverlay=rpi-digiampplus

[cm4]
otg_mode=1

[cm5]
dtoverlay=dwc2
dtparam=dr_mode=host
dtoverlay=

[all]
initramfs initrd followkernel
```

I am not currently using an encrypted root. I am trying to boot a 6.12 kernel.

I have also tried disabling vc4-kms-v3d, and then accessing the pi over the network, but I don't see it appear on the network.

Answering the questions as best as I can:

*  Did you try booting the unchanged rpi5-installer image provided by the flake?
  * I haven't tried this yet, apologies.
  * I'm booting a configuration I have (approximately) used before but porting it to use this repository.
*  Does it boot without rpi-digiampplus overlay?
  * I have tried removing this and it doesn't appear to help.
*  Has this particular nixos system ever booted on RPi5? Maybe there're some changes that you've made that could have broken it?
  * Yes, I was able to bot on a 6.6 kernel using an older configuration. But I wasn't able to get myself into a position where I could `nixos-rebuild boot` to update the kernel, which I'm hoping to achieve with the nixos overlay from this repository.
*  May you could provide more complete boot log?
  * Unfortunately not immediately, apologies. I don't have a uart to hand. I can try harder if it is likely to help but it might take me a bit of time to get around to it (and it might be a photo of a screen...)

Other potential oddities, I have these set:

```
  boot.loader.raspberryPi.enable = true;
  hardware.raspberry-pi.config.all.dt-overlays.rpi-digiampplus.enable = true;
  hardware.raspberry-pi.config.all.base-dt-params = {
    pciex1 = {
      enable = true;
      value = "on";
    };
    pciex1_gen = {
      enable = true;
      value = "3";
    };
  };

  # boot.kernelPackages = (import nixpkgs { system = "x86_64-linux"; crossSystem = { system = "aarch64-linux"; }; }).linuxPackages_latest;
  boot.kernelPatches = [ {
    name = "realtime-config";
    patch = null;
    extraConfig = ''
      PREEMPT_VOLUNTARY n
      PREEMPT_RT y
      CPU_FREQ_GOV_INTERACTIVE n
      FAIR_GROUP_SCHED n
      CPU_FREQ_TIMES n
    '';
  } ];
```
  

My flake config fragment looks something like this:

```
nixosConfigurations.pi = nixos-raspberrypi.lib.nixosSystem {
      system = "aarch64-linux";
      # specialArgs = { inherit nixpkgs nixos-raspberrypi; };
      specialArgs = { inherit nixos-raspberrypi; };
      modules = [
        nixos-raspberrypi.nixosModules.raspberry-pi-5.base
        nixos-raspberrypi.nixosModules.sd-image

        ./pi/configuration.nix
```

I have an nvme drive, but I'm trying to boot from an sd card right now. Once I'm booted off the sd card I intend to install to the nvme.

I can try booting from the rpi5-installer if you think this will help.

I have identified that my raspberry pi 5 is a D0 stepping of the broadcom chip, in case that's relevant, I did see some discussion about uboot not yet supporting this, but I believe I'm not using uboot yet(?).

pwaller, 2025-07-20:
> I see the config.txt differs from the default one, and also appears to be manually edited

You caught me! I was attempting to make the config.txt look more like the one that is known booting for me.

nvmd, 2025-07-23:
> I have identified that my raspberry pi 5 is a D0 stepping of the broadcom chip, in case that's relevant, I did see some discussion about uboot not yet supporting this, but I believe I'm not using uboot yet(?).

That shouldn't be a problem. `kernel` boot is used for RPi5, unless you've specifically set `boot.loader.rasbperryPi.bootloader = "uboot"`.

> I can try booting from the rpi5-installer if you think this will help.

I think it will, at least we'll know that the very minimal configuration works on your system. After verifying it works, that could be a good starting point to migrate your existing configuration to, seeing which change will eventually break the system, for example.

So what is your current situation, exactly? It's actually quite hard to understand right now what's going on.
Are you generating the `sd-image` and trying to boot it?
Or are you migrating your existing configuration from some other infrastructure (which one?) to `nixos-raspberrypi`? If this is the case, you don't really need to import `nixos-raspberrypi.nixosModules.sd-image`, you can just configure `fileSystems` like this: https://github.com/nvmd/nixos-raspberrypi-demo/blob/main/flake.nix#L205-L223

ryand56, 2025-07-24:
> Same here while running `nixos-rebuild test`. When I try `nixos rebuild switch` I get `cp: error writing '/boot/firmware/bcm2708-rpi-b.dtb.tmp': No space left on device`.

I had this issue before but I ran `sudo nix-collect-garbage -d`

nevesenin, 2025-07-25:
> I had this issue before but I ran sudo nix-collect-garbage -d and deleted old kernels in /boot

Yes, I did that before. I set up a bunch of Raspberry Pis with https://github.com/nix-community/raspberry-pi-nix, which have a default FIRMWARE partition size of 128MB. That's just too small, especially for future features like preserving build generations.
To deal with that, I'll create backups of the partitions, repartition the disk and restore the partitions. Or just reinstall the system. If anyone knows a more convenient way, please let me know.
To some of the before-mentioned Raspberrys I have remote access only, so by now I feel I can't move away from raspberry-pi-nix with them.

pwaller, 2025-07-26:
> > can try booting from the rpi5-installer if you think this will help.
> 
> I think it will, at least we'll know that the very minimal configuration works on your system.

Unfortunately, I booted `nvmd/nixos-raspberrypi/31a4d34a966522578133a616333c5e431d22032b#nixosConfigurations.rpi5-installer`, and this hangs at exactly the same point. My previous use of an alternate kernel also conclusively shows that I am booting the correct kernel since the console shows 'Linux 6.6.51' and not my customized 6.12.

So something else must be going on.

The last messages (OCR'd from a photo, which made a bit of a mess of it) are:

```
[3.480961] Freeing unused kernel memory: 2496K
[3.485965] Run /init as init process
[3.519427] mmc1: new ultra high speed DDR50 SDIO card at address 0001
[3.527033] device-mapper; ioctl: 4,48,0-joctl (2023-03-01) initialised: dm-deuel@redhat.com
[3.618708] mmc0: new ultra high speed DDR50 SDHC card at address aaaa
[3.631264] mmcblk®: mmc0:aaaa SS32G 29.7 GIB
[3.631992] macb 1/00100000, ethernet endo: renamed from etho
[3.644342] mmcblk0: p1 p2
[3.649580] mmcblk0: mmc0:aaaa SS32G 29.? GiB (quirks 0x00004000)
[3.673558] input: Microsoft Wired Keyboard 600 as /devices/platform/axi/1000120000.pcie/1f0
[3.707863] vc4-drm axi:gpu: bcm2712_iommu_of_xlate: MMU 1000005200. iommu
```

This vc4-drm message is the same one as I had with the 6.12 kernel.

nvmd, 2025-07-26:
So do you use the _unmodified_ `rpi5-installer` or not? The default version of the kernel in use is `v6_12_25`, not 6.6.51.

https://github.com/nvmd/nixos-raspberrypi/blob/157cf84d037896411b2d2b3fde3d03a8967d8aed/overlays/linux-and-firmware.nix#L30-L52

pwaller, 2025-07-26:
Ah, I wonder if the linux version I saw in my log is a red herring: it looks like I was reading the line  `usb usb4: Manufacturer: Linux 6.6.51 xhci-hcd`.

If I dump the kernel.img from FIRMWARE partition, then I see from strings `6.12.25` as expected:

```
Linux version 6.12.25 (nixbld@localhost) (gcc (GCC) 14.2.1 20250322, GNU ld (GNU Binutils) 2.44) #1-NixOS SMP PREEMPT Tue Jan  1 00:00:00 UTC 1980
```

I did a literal `nix build github:nvmd/nixos-raspberrypi/31a4d34a966522578133a616333c5e431d22032b#nixosConfigurations.rpi5-installer.config.system.build.image` to get the image.

pwaller, 2025-07-27:
Looks like others hit this:

* https://github.com/nix-community/raspberry-pi-nix/issues/123
   * Dated Feb 14th this year - no resolution - but 0.4.1 tag worked for them.
* https://hub.mender.io/t/boot-stopping-after-four-berries/7256/10
   * Removing `dtoverlay=vc4-kms-v3d` helped (It didn't help me unfortunately, as in my original report)

pwaller, 2025-07-27:
Full set of commits not in nix-community/raspberry-pi-nix@v0.4.1: 
I note there isn't much there, but there are some kernel bumps.

`$ TZ=UTC git log --date=format:%Y-%m-%d --format='* nix-community/raspberry-pi-nix@%H %cd %s' --no-merges  v0.4.1..`
* nix-community/raspberry-pi-nix@abd2352d0405bdae6c43d022e4c65085311d5d7d 2025-03-01 bump 6.6.y and 6.12.y kernel versions
* nix-community/raspberry-pi-nix@b54486ef66721f3e6c8fad98c83b8e7f61e2aad4 2025-02-17 use nixpkgs upstream uboot
* nix-community/raspberry-pi-nix@824ce252585a8fbf8b50023da67c8405d5f4ff85 2025-02-08 Make inclusion of console=serial0 in kernelParams (cmdline.txt) configurable
* nix-community/raspberry-pi-nix@36d2ee3511581cbcb58bb09a820195c6da72408b 2025-02-04 Automatically update `flake.lock` using GitHub Actions
* nix-community/raspberry-pi-nix@25118248489e047a7da43a21409b457aa2af315e 2025-01-29 Ignore kernel configuration errors
* nix-community/raspberry-pi-nix@e94b747760d09619786a64b84fad30fd6839dd4f 2025-01-29 Drop kernel 6.10.12 in favour of 6.12.11
* nix-community/raspberry-pi-nix@85faa0f775e3590ff4223c80786862c6f26018c5 2025-01-08 matching firmware
* nix-community/raspberry-pi-nix@340fc9e3806bd789418a815f2181d7d6be9045d7 2025-01-08 default to stable kernel
* nix-community/raspberry-pi-nix@25b46a0be002fbc4f4b5b36ba7d2ce9652d23c53 2025-01-08 add stable kernel
* nix-community/raspberry-pi-nix@f85fbd621f353db2654173300c75e8c00fc8b6b5 2025-01-06 bump nixpkgs to 24.11
* nix-community/raspberry-pi-nix@97c51868c6a820951439162507b80e1b026d8ad4 2024-12-26 Update kernel from 6.6.54 to 6.6.67
* nix-community/raspberry-pi-nix@4a74a0208c3ab31e4e7d07367361aef2cf648b1a 2024-11-26 Determine partition number for root partition by PARTN column
* nix-community/raspberry-pi-nix@aaec735faf81ff05356d65c7408136d2c1522d34 2024-11-18 add stability note
* nix-community/raspberry-pi-nix@848c5add7f5c4ad0792c86d9db620c6aa4abc30e 2024-11-12 readme tweaks
* nix-community/raspberry-pi-nix@b7c3578710785c93c7d9e714bb1c8dd2421875bb 2024-11-12 move comment
* nix-community/raspberry-pi-nix@beb01f006a795fa1c3d46ca2a7a3dd199d919b0b 2024-11-11 add firmware-partition-label option
* nix-community/raspberry-pi-nix@61d43e92fd7530d0162d9125a6cb9f27b4e1e32c 2024-11-04 use config.system.boot.loader.kernelFile
* nix-community/raspberry-pi-nix@f872d7748f503af66ed577aaf9b621c69235f40c 2024-11-03 shuffle kernel params
* nix-community/raspberry-pi-nix@ee53143215d82acfe159009b5745795828fddb5b 2024-11-03 add initrd
* nix-community/raspberry-pi-nix@d039a262cecc42897c4d8b3014e83cadbd9c46e3 2024-11-03 disable sd-image

nvmd, 2025-07-28:
Hi @pwaller. Thank you for the research!

While I'm looking into it, could you please give me a some details about your hardware:
- What is the model of RPi5 you're using?
- Do you have the latest EEPROM version installed?
- What is the EEPROM configuration accessible with `rpi-eeprom-config --edit`?

I also understand you're mainly using nvme drive. Right now, `nixos-raspberrypi`-based systems doesn't boot for you neither from sd-card, nor from nvme, do I understand correctly?

> * I've seen reports on the net about graphics being broken, which made me wonder if the pi with nvmd/nixos-raspberrypi is actually booted up but without graphics. But I can't see evidence of it coming onto the network either and it's not responding to ping on the expected ip address.

I'm personally running RPi5 8Gb with graphics and nvme drive using linux 6.6.31. So graphics don't seem to be the only culprit. I do have FIRMWARE on sd-card, though. I.e. it starts booting from sd-card, and then proceeds with the rest of the system from nvme drive.
I did, however, have the same issue with LUKS password not showing up when `vc4-kms-v3d` is enabled, as I mentioned here  https://github.com/nvmd/nixos-raspberrypi/issues/49#issuecomment-3094621094

nvmd, 2025-07-28:
@pwaller @ryand56 

> 2\. My existing nvme boot from my old configuration is still working,

What is your old/existing configuration is? Is it using `nixos-hardware` or `raspberry-pi-nix`? Maybe you could provide some relevant portions from it?

I think analyzing the differences between your existing working config and this flake that currently doesn't work for you is the key to find the solution?

pwaller, 2025-07-28:
> Right now, nixos-raspberrypi-based systems doesn't boot for you neither from sd-card, nor from nvme, do I understand correctly?

While I have an existing functioning nvme boot, I'm leaving it alone with the previous config on it until I have a functioning boot. So, unknown, with regards to nvme.

> What is your old/existing configuration is?

The functioning setup is based on nixpkgs `25.05.20241219.d70bd19`, nixos/nixpkgs@d70bd19e0a38ad4790d3913bf08fcbfc9eeca507 nix-community/raspberry-pi-nix@3e8100d5e976a6a2be363015cb33463af9ef441a.

I used modules:

```nix
   raspberry-pi-nix.nixosModules.raspberry-pi
   raspberry-pi-nix.nixosModules.sd-image
```

and settings:

```nix
  raspberry-pi-nix.board = "bcm2712";
  boot.kernelPatches = [ {
    name = "realtime-config";
    patch = null;
    extraConfig = ''
      PREEMPT_VOLUNTARY n
      PREEMPT_RT y
      CPU_FREQ_GOV_INTERACTIVE n
      FAIR_GROUP_SCHED n
      CPU_FREQ_TIMES n
    '';
  } ];
```
  
> * What is the model of RPi5 you're using?

16GiB.

> * Do you have the latest EEPROM version installed?

`rpi-eeprom-update` reports:

```
BOOTLOADER: up to date
   CURRENT: Tue Nov 12 04:10:44 PM UTC 2024 (1731427844)
    LATEST: Tue Nov 12 04:10:44 PM UTC 2024 (1731427844)
   RELEASE: default (/nix/store/qgmlsl8q1jmzal24z4qg3mhl3s74gysi-raspberrypi-eeprom-2024.11.12-2712/lib/firmware/raspberrypi/bootloader-2712/default)
            Use raspi-config to change the release.
```

I will try flashing `pieeprom-2025-05-08.bin` which is the latest I see in nixos-unstable.

> * What is the EEPROM configuration accessible with rpi-eeprom-config --edit?
```
[all]
BOOT_UART=1
POWER_OFF_ON_HALT=0
BOOT_ORDER=0xf466
PCIE_PROBE=1
```

---

As an additional experiment, I took my firmware/config from the functioning setup, and copied it to the nonfunctioning setup (taking care to preserve `cmdline.txt`, `kernel.img` `initrd` and `nixos-kernels/`).

As another point of reference, the functioning setup has Linux `6.6.51` (compared with 6.6.31 for my current non-functioning setup).

I lack a uart, so I am relying on what the monitor shows and whether I ping it over the network. At the moment if the boot does not proceed past `[3.707863] vc4-drm axi:gpu: bcm2712_iommu_of_xlate: MMU 1000005200. iommu` for more than 5-10 seconds then I consider the machine hung at boot. I'm not totally confident this is correct, the machine could be doing something behind that but I can't tell. It doesn't respond to sysrq (actually I'm not sure if this is swithced on), but it does respond to ctrl-alt-del by rebooting.

nvmd, 2025-07-30:
> BOOT_ORDER=0xf466

So now the boot order is NVME->NVMe->USB (https://www.raspberrypi.com/documentation/computers/raspberry-pi.html#BOOT_ORDER).
Are you booting the `sd-image` from USB drive, or do you change the boot order to include `0x1` for SD card device, when you're doing the experimentation? Are you disconnection the NVMe drive?

I don't remember from the top of my head right now how does it exactly work, but the problem could theoretically be: once the FIRMWARE from `sd-image` boots, it proceeds with `init` not from the nix store on the `sd-image`, but from the `nix` store on your NVMe.
Or vice versa, if you don't disconnect the NVMe, its FIRMWARE could get loaded first, subsequently picking up the wrong nix store, making you the impression that SD image doesn't work.
That, of course, depends on how exactly the “bootloader” discovers the correct nix store to proceed. I don't remember that right now, but I believe this direction is worth exploring, ruling out this scenario.

pwaller, 2025-08-01:
I think you've nailed it with your amazing diligence. Unseating the nvme cable has fixed the boot hang. It did not occur to me that the boot order would be booting the nvme first. Now I have a confirmed working boot with 6.6.31. Part of the issue there is that the monitor is coming on too late to observe what is happening in early boot.

pwaller, 2025-08-01:
I discovered an additional problem which might be messing things up even once the boot order is fixed. My nvme drive has a partition with the same UUID and partition label as on the sdcard, as shown by `blkid`.

Consequently I think there may be non-deterministic mount confusion happening. I have just observed one boot fail and another boot succeed. At least now I know this is going on, I can arrange to do a proper nixos-install onto the nvme drive from the sd card boot and arrange that the root partition on the nvme has a properly unique uuid and label.