nvmd/nixos-raspberrypi #60: Modular Generational Boot Loader

nvmd, 2025-07-21:
_This is a work-in-progress design document and a tracking issue for the implementation of the modular generational boot loader, aimed at fixing limitations and shortcomings of the existing legacy implementation inherited from `nixpkgs`'s `boot.loader.raspberryPi`.
Comments, suggestions and contributions are welcome._

## Tasks/Sub-issues:
- `kernel` (kernelboot2) generational bootloader (currently available for testing in [`usable-kernelboot-gens`](https://github.com/nvmd/nixos-raspberrypi/tree/usable-kernelboot-gens))
- vendor kernel overlays must include README for nixos generation to be detected as valid by the raspberry firmware (firmwarePackage already includes it) – see https://www.raspberrypi.com/documentation/computers/config_txt.html#os_prefix, https://www.raspberrypi.com/documentation/computers/config_txt.html#overlay_prefix
- Support for shared `overlays/` – separate configuration option? “Unless ${os_prefix}${overlay_prefix}README exists, overlays are shared with the main OS (i.e. os_prefix is ignored).” – I guess that implies, that if `overlays/` are missing in ${os_prefix}, than ${os_prefix} will be ignored as well, i.e. that’s the same shared scenario
- Develop enduser migration instructions to the generational bootloader (kernel)
- To reduce closure size when kernel’s dtbs and overlays are used (`useGenerationDeviceTree=true`), split raspberrypifw into raspberry-firmware and raspberry-dtbs

# Modular Generational Boot Loader

## Limitations and shortcomings of the existing implementation of the bootloader infrastructure
- Kernel, DTBs and overlays need to be matched to the kernel, but `FIRMWARE` can store only single set of them, i.e. making rollback potentially impossible due to DTB mismatch even then the kernel or uboot binary of the previous generation can be recovered
- `uboot`: Only the latest uboot binary is stored
- `kernelboot`: Effectively only the latest kernel is stored. `kernelboot` stores multiple kernel,initrd,cmdline.txt versions, but they’re unusable because the implementation didn’t take into account that FAT-32 formatted partition doesn’t support symlinks, desired generation name can hardly be matched to the kernel binary file with a cryptic name

## Design goals of the proposed solution
- Minimize footprint of the bootloader infrastructure not covered by nixos generations, "non-generational/generation-independent" part
- Maximum possible nixos generation separation for a number of nixos generations up to a set limit
- Improve usability of the kernel bootloader, the only fully-supported option for RPi5
- Improve customizability, provisioning of additional “firmware” files, e.g. overlays if needed by the end-user
- Improve maintainability of the codebase

## Implementation approach
Separate nixos generations (generation-dependent files) are stored in a `nixos/<nixos-generation-name>/`.
Generation to be loaded by the Raspberry’s firmware is controlled by the `os_prefix` option in `config.txt`.

`FIRMWARE=/boot/firmware` by default, configurable with `boot.loader.firmwarePath` option.

Generation-independent files -> `${FIRMWARE}/`
- “boot code”
- config.txt

Generation-dependent files -> `${FIRMWARE}/nixos/<nixos-generation-name>/`
- kernel:
    - kernel.img
    - initrd
    - cmdline.txt
    - kernel-link
- uboot:
    - u-boot-rpi-arm64.bin or u-boot-rpi.bin
- kernel and uboot:
    - system-link
    - DTBs
    - overlays (when generation-specific overlays are used. Overlays can also be shared, even though it’s not clear how they’re supposed to be always? compatible with DTBs which can’t be shared)

Default value for the `os_prefix=nixos/default`. User can temporarily boot the desired nixos generation by setting the option appropriately, e.g. `os_prefix=nixos/103-default/`.
These changes to `config.txt` will be overwritten when nixos configuration is switched with `nixos-rebuild switch`.

## Limitations and shortcomings of the proposed implementation of the bootloader infrastructure
- Increased storage space usage of the `FIRMWARE` partition. To keep the NixOS generations as independent as possible the generation-dependent files are need to be copied and kept for each generation. No symlinks or hardlinks are allowed. Mitigated by the garbage collection mechanism controlled by the newly-introduced option `kernelbootConfigurationLimit`

## Implementation details
Modules:
- `nixos-generations-builder`
    - Calls installFirmwareBuilder to install generation-independent files
    - Calls nixosGenBuilder to install generation-dependent files, depends on the boot method: kernel, uboot
    - Performs garbage collection of the obsolete nixos configurations above `kernelbootConfigurationLimit`
    - Installs nixos generations:
        - Lazily, to extend life of sd-cards
        - Atomically, to reduce risks associated with interruptions of the installation script
- `nixosGenBuilder` => `kernelboot-gen-builder`, `uboot-gen-builder`
    - Installs generation-dependent files
        - Boot method-specific files
        - Calls installDeviceTree to install generation-dependent device tree and overlays
- `installDeviceTree`
    - Installs device tree files and overlays either from:
        - `firmwarePackage`, `pkgs.raspberrypifw` by default
        - dtbs and overlays of the kernel of the nixos generation
- `installFirmwareBuilder`
    - Installs generation-independent files:
        - “Boot code”
        - `config.txt`

## Migration instructions
- Ensure `FIRMWARE` partition is big enough, set `kernelbootConfigurationLimit` appropriately, single nixos generation consumes about ~XXXMB
- Backup existing `/boot/firmware` `cp -r /boot/firmware /boot/firmware.bkp` Backup should be accessible in case you’re not able to boot the system anymore – you’ll be able to replace the FIRMWARE partition contents with the backup you’ve just made.
- Cleanup/Remove existing /boot/firmware `rm -rf /boot/firmware/*`
- Set `boot.loader.raspberryPi.bootloader = "kernelboot2”`;
- Switch to configuration
- Inspect newly configured /boot/firmware, should look something like `tree /boot/firmware   <output…>`. `config.txt` must contain `os_prefix=`, directory should contain its value

## Deployment plan
- Merged into `develop` as an “opt-in”
- Merged into `main` as an “opt-in”
- Deprecation notice for `kernelboot` in `main`, instructions to migrate to `kernelboot2`
- `kernelboot` is deprecated
- `kernelboot2` renaming to `kernel` notice
- `kernelboot2` renamed to `kernel`

tobiasBora, 2025-11-30:
Seems like main has been rebased recently on develop, so I guess `kernel` is available in main now?

nvmd, 2025-11-30:
It was available in main already since the previous release as an opt-in. Now, only the deprecation warning for `kernelboot` was added.