nvmd/nixos-raspberrypi #66: develop installerImages.rpi5 is broken

PulfordJ, 2025-07-25:
First off thanks for this great project I didn't realise pi5 was less supported by NixOS than the prior releases!

`[pi@media-center-pi:~/nixos-raspberrypi]$ nix build github:nvmd/nixos-raspberrypi/main#installerImages.rpi5`

main works fine, tested after I found  found nvmd suggested to use main going forward here: [](https://github.com/nvmd/nixos-raspberrypi/issues/30#issuecomment-2940139440).

```
[pi@media-center-pi:~/nixos-raspberrypi]$ nix build github:nvmd/nixos-raspberrypi#installerImages.rpi5
error: path '/nix/store/bm1kn36lrhvzp5d1w9rql4jvhs27jcky-linux_rpi-bcm2712-6.12.25-stable_20250428-modules-shrunk/lib' is not in the Nix store
```

Not a major issue, main is more than enough for me to work with just thought I'd highlight it.

nvmd, 2025-07-25:
Hi @PulfordJ. Thank you for the report.
I haven't been so far able to reproduce the issue you have, though

PulfordJ, 2025-07-25:
fascinating, I wonder if you a stale copy of the path, did you try garbage collecting first?

PulfordJ, 2025-07-28:
Further notes in the linked issue above, feel free to close cheers!