nvmd/nixos-raspberrypi #67: Thoughts on matching sd-image-aarch64.nix so one can build an installer image or switch all in one repo off of a specific configuration

PulfordJ, 2025-07-25:
First of all I want to say I love this project and seeing how active you are!

Having come from a pi3 I do miss having sd-image-aarch64.nix which, if imported into my nixosConfiguration, gave me an entrypoint to build an image containing my nixosConfiguration. Alas ofc sd-image-aarch64.nix does not support pi5.

Example config:

```
    nixosConfigurations = {
      media-center-pi = nixpkgs.lib.nixosSystem {
        specialArgs = {inherit project_root inputs outputs;};
	system = "aarch64-linux";
        modules = [
	"${nixpkgs}/nixos/modules/installer/sd-card/sd-image-aarch64.nix"
	{
	nixpkgs.config.allowUnsupportedSystem = true;
	nixpkgs.hostPlatform.system = "aarch64-linux";
	nixpkgs.buildPlatform.system = "aarch64-linux";
	}
          # > Our main nixos configuration file <
          ./nixos/configuration.nix
        ];
      };
    };

    images.rpi3 = nixosConfigurations.media-center-pi.config.system.build.sdImage;
```
Example output:

```
# build the image
nix build .#images.rpi3 
# just apply the configuration ontop
sudo nixos-rebuild switch  --flake .#media-center-pi --target-host pi@media-center-alias --use-remote-sudo
```

Ideally there'd be some way for me to do the same with nixos-raspberrypi. It could also negate some of the grub misconfiguration issues other users are facing by enforcing that certain settings are not set when building the configuration too. I'd be happy to help, I'm just looking to get your thoughts on how feasible this would be to add / maybe there's some horrible hurdles I haven't considered.

PulfordJ, 2025-07-25:
I totally understand there's a custom-user-config setting for the sdimage installers btw. It just seems like the dependencies are the wrong way round. Ideally an arbitrary configuration can import something that builds the image under the hood rather than an image importing the config as a parameter.

nvmd, 2025-07-25:
Hi!

> Having come from a pi3 I do miss having sd-image-aarch64.nix which, if imported into my nixosConfiguration, gave me an entrypoint to build an image containing my nixosConfiguration. Alas ofc sd-image-aarch64.nix does not support pi5.

I believe `nixos-raspberrypi.nixosModules.sd-image` module does what you're looking for.

> I totally understand there's a custom-user-config setting for the sdimage installers btw. It just seems like the dependencies are the wrong way round. Ideally an arbitrary configuration can import [...]

These aren't actually mean to be consumed directly. `custom-user-config` is there merely to organize the code of the installers in a little more comprehensible way without duplication and to show which part of the configuration is required for the flake, and which one is not.

nvmd, 2025-07-30:
Ok, I'm closing this issue for now, then. Should issues arise or should you have further questions, please open an issue or create a new discussion!

closed 2025-07-30 by nvmd