nvmd/nixos-raspberrypi #69: Hang on `EFI stub: Exiting boot services...`

crabdancing, 2025-07-26:
I am trying to setup an encrypted ZFS boot via [UEFI for the RPi4](https://github.com/pftf/RPi4). I've tried _many_ different things, and I always end up stuck on this same problem -- with the main factor in common seeming to be native ZFS encrypted boot. I've tried both using built-in boot management that comes with raspberry-pi, and also overriding it:

`uefi-boot.nix`:

```nix
{lib, ...}: {
  # Override the Pi bootloader
  boot.loader.raspberryPi.enable = lib.mkForce false;

  # Enable standard UEFI boot
  boot.loader.efi.canTouchEfiVariables = true;
  boot.loader.systemd-boot.enable = true;
  boot.loader.grub.enable = false;

  boot.initrd.availableKernelModules = ["xhci_pci" "usbhid" "uas" "usb_storage"];
  # boot.loader.grub.enable = false;

  # boot.loader.efi.canTouchEfiVariables = true;
  # boot.loader.systemd-boot.enable = true;
  boot.loader.generic-extlinux-compatible.enable = lib.mkForce false;
}
```

If it seems like I'm throwing everything at the wall and seeing what sticks, that's because it's exactly what my methodology has devolved to at this point. :P

`disko.nix`:

```nix
{
  disko.devices = {
    disk = {
      main = {
        type = "disk";
        device = "/dev/disk/by-id/ata-Samsung_SSD_860_EVO_M.2_250GB_S5GFNS0N807358R";
        content = {
          type = "gpt";
          partitions = {
            ESP = {
              size = "5G";
              type = "EF00";
              # attributes = [2];
              content = {
                type = "filesystem";
                format = "vfat";
                mountpoint = "/boot";
                mountOptions = [
                  "noatime"
                  "noauto"
                  "x-systemd.automount"
                  "x-systemd.idle-timeout=1min"
                  "umask=0077"
                ];
              };
            };
            zfs = {
              size = "100%";
              content = {
                type = "zfs";
                pool = "rpool";
              };
            };
          };
        };
      };
    };
    zpool = {
      rpool = {
        type = "zpool";

        options = {
          ashift = "12";
          autotrim = "on";
        };

        rootFsOptions = {
          atime = "off";
          xattr = "sa";
          acltype = "posixacl";
          normalization = "formD";
          dnodesize = "auto";
          mountpoint = "none";
          canmount = "off";
        };

        datasets = {
          "enc/data" = {
            type = "zfs_fs";
            options = {
              mountpoint = "none";
              "com.sun:auto-snapshot" = "true";
            };
          };
          "enc/nix-store" = {
            type = "zfs_fs";
            options = {
              mountpoint = "legacy";
              "com.sun:auto-snapshot" = "false";
            };
            mountpoint = "/nix/store";
          };
          "enc/root" = {
            type = "zfs_fs";
            options = {
              mountpoint = "legacy";
              "com.sun:auto-snapshot" = "true";
            };
            mountpoint = "/";
          };
          "enc" = {
            type = "zfs_fs";
            options = {
              mountpoint = "none";
              encryption = "aes-256-gcm";
              keyformat = "passphrase";
            };
          };
        };
      };
    };
  };
}
```

For clarity, I'm trying to setup the boot from a microSD with the UEFI blobs extracted to it, then from there, it's meant to load a UEFI-bootable filesystem on a SATA device via an ArgonONE case adapter.

This setup has gotten me as far as UEFI boot screen -> generation selection, but gets stuck with the following being the full contents of my screen:

```
EFU stub: Booting Linux Kernel...
EFI stub: Loading initrd from LINUX_EFI_INITRD_MEDIA_GUID device path
EFI stub: Generating empty DTB
EFI stub: Exiting boot services...
```

Does anyone have any ideas?

nvmd, 2025-07-30:
Hi @crabdancing!

It's not very clear how exactly are you using this flake? Strictly speaking, `uefi` boot isn't supported by the project, but are you, maybe, using some component of it?
I see `boot.loader.raspberryPi.enable` is disabled, which is definitely the right thing to do, but nothing else related.

I've been experimenting with UEFI with RPi4+ArgonONE some time ago myself. But, frankly speaking, I don't quite remember if I managed to get it working even without such "complications" as zfs with encryption.
I feel like I've given up back then, more or less. So can't provide any guidance even on the rights of the "off-topic" discussion.

crabdancing, 2025-07-30:
Oh, yeah. Um, I was just calling this flake's `lib.nixosSystem` and passing it the full suite of hardware support modules.

Also, no worries. I already got it working, using the stock kernel. (without a special flake -- just upstream nixpkgs) Through a lot of hairpulling, I figured out a number of odd behaviors / failure modes.

I'm using `_latest` just fine, though! So something about the vendor kernel _really_ doesn't like being booted in that config. I would have looked into it more, but I kinda ran out of time. You _can_ get it _kind of_ working (as in, you're at least in multiuser space) with this DTB config:

```nix
hardware.deviceTree = {
  enable = true;
  name = "broadcom/bcm2711-rpi-4-b.dtb";
  package = lib.mkForce "${config.boot.kernelPackages.kernel}/dtbs";
};
```

If you _don't_ do this, and you're on the `_rpi4` kernel, I believe it does this:

```
EFI stub: Generating empty DTB
EFI stub: Exiting boot services...
```

But if you look try the 'staple' solution:

```nix
hardware = {
  raspberry-pi."4".apply-overlays-dtmerge.enable = true;
  deviceTree = {
    enable = true;
    filter = "*rpi-4-*.dtb";
  };
};
```

you get this:

```
FileNotFoundError: [Errno 2] No such file or directory: '/nix/store/w97700g1kngv5sf5jj379m60byphcw0i-linux_rpi-bcm2711-6.12.25-stable_20250428/dtbs/bcm2711-rpi-4-b.dtb'
Failed to install bootloader
```

Anyways, I've given up on using the `_rpi4` kernel unless I see some material benefits, because it looks like the stock `_latest` aarch64 kernel can interact with all the same hw just fine (I haven't tried GPIO yet though :/ )

The way I've been doing it is to use this flake to bake a rpi microSD, which I then bootstrap off of onto a SATA drive over SSH, similar to my ISO SSH bootstrap process for conventional x86_64 boxes.