nvmd/nixos-raspberrypi #78: Move to NixOS-Hardware

1nv0k32, 2025-08-05:
Hi there,

Would it be possible to upstream this awesome work to [nixos-hardware](https://github.com/NixOS/nixos-hardware)?

JamieMagee, 2026-04-28:
Quick update. Three more PRs are open against nixos-hardware, all aimed at making the existing raspberry-pi kernel package usable as a foundation:

- [#1839](https://github.com/NixOS/nixos-hardware/pull/1839) replaces deprecated `lib.overrideDerivation` with `.overrideAttrs`. The old form silently drops `passthru`, which breaks downstream customization.
- [#1840](https://github.com/NixOS/nixos-hardware/pull/1840) fixes the vendor -> upstream DTB copy in `postFixup`. It was gated on host arch instead of `rpiVersion`, so Pi 4+ was getting the workaround it doesn't need. Also fixes `bcm2835-rpi-zero.dtb` (copied from the wrong board) and drops a dead `bcm2838-rpi-4-b.dtb` copy nothing references.
- [#1841](https://github.com/NixOS/nixos-hardware/pull/1841) adds `structuredExtraConfig` so nixpkgs `common-config.nix` stops overriding vendor defconfig values. This also kills the old sed hack on `.config`/`auto.conf`.

With those merged, the upstream kernel matches what this flake builds closely enough that @pta2002's binary cache concern stops being a blocker. Nothing prevents this repo (or a sibling project) from publishing builds of the nixos-hardware kernel.

On the bootloader question, I have to walk back what I said earlier. I went and actually checked u-boot upstream and it's worse than I thought.

openSUSE [shipped Tumbleweed support for Pi 5 in November 2025](https://news.opensuse.org/2025/11/04/raspberrypi5-opensuse/), using `rpi_arm64_defconfig` then GRUB2 then kernel from SD. So yes, it boots. But on Pi 5 every interesting controller (USB, Ethernet, WiFi) sits behind the RP1 chip over PCIe, and U-Boot has no BCM2712 PCIe driver. I checked u-boot master and `drivers/pci/pcie_brcmstb.c` still only matches `brcm,bcm2711-pcie`. The [BCM2712 PCIe series Ivan Ivanov posted to the list](https://lore.kernel.org/u-boot/20251010161442.410C4227AAE@verein.lst.de/) back in October has been sitting there for six months without being merged.

So inside U-Boot/GRUB on a Pi 5 right now: SD only. No USB keyboard at the GRUB prompt, no USB boot, no PXE. Linux brings up PCIe once it takes over, so userspace is fine, but the bootloader stage is genuinely degraded compared to the firmware-direct-kernel path this flake uses. That'd be a noticeable regression to ship to users.

So I think we keep the direct-boot path for Pi 5 specifically until U-Boot catches up. Pi 2/3/4 are a different story. extlinux works fine on those today, and nixos-hardware should be able to cover them once `config.txt` and firmware install land.

After these three PRs merge I'll wire `hardware.raspberry-pi.configtxt.settings` ([#1788](https://github.com/NixOS/nixos-hardware/pull/1788)) into the board profiles and start on the firmware install pieces.

pta2002, 2026-04-29:
@JamieMagee thanks for this write-up, I agree that those changes are very important! However, one question:

> With those merged, the upstream kernel matches what this flake builds closely enough that @pta2002's binary cache concern stops being a blocker. Nothing prevents this repo (or a sibling project) from publishing builds of the nixos-hardware kernel.

What do you mean? Is the upstream kernel now good enough to run the Pi 5 without issues? Because since those changes are for nixos-hardware's defconfig, then they would force a recompilation anyway which takes several hours on the Pi without a cache.

1nv0k32, 2026-09-06:
> Pi 5 NVMe is no longer a blocker. U-Boot's NVMe DMA fix has landed for the 2026.10 release. [I have cherry-picked the series pieces and Raspberry Pi config changes onto nixpkgs' U-Boot 2026.07, and I am daily driving a Pi 5 booting directly from NVMe with nixos-hardware](https://github.com/JamieMagee/nix-config/blob/main/overlays/uboot-rpi-arm64.nix). Once nixpkgs updates U-Boot and carries those config changes, this repository's custom direct firmware boot path should no longer be needed.

@JamieMagee 
I had a really weird issue with the NVMe and Uboot 2026.07 with those patches you are using. Uboot logo was showing and nothing was happening. After a bit of digging, I've made it kinda work with below config.
https://codeberg.org/1nv0k32/nixconf/src/branch/main/system/rpi5/default.nix#L72-L74
If you have any clue what is going on, please give me a hint 😅

nvmd, 2026-09-06:
@1nv0k32 Even though I appreciate general efforts of bringing the uboot to work as well as lots of other things, let's maybe keep this discussion focused on the issues of this particular flake instead of something that happens somewhere else on the internet.

JamieMagee, 2026-09-06:
@1nv0k32 I found a missing bootable flag on ROOT in your Disko config. U-Boot considers the firmware ESP bootable, but its automatic scan skips ROOT. That's where `/boot/extlinux/extlinux.conf` lives, so it never gets read. Your explicit `nvme 0:2` loads bypass that.

I'd set the flag in Disko:

```nix
disko.devices.disk.nvme0.content.partitions.root.attributes = [ 2 ];
```

For the existing disk, confirm partition 2 is ROOT, then run

```
sudo sgdisk --attributes=2:set:2 /dev/nvme0n1
```

That sets GPT attribute bit 2 without reformatting anything. A rebuild alone won't update the partition table.

I also noticed `sysboot` in [your earlier commits](https://codeberg.org/1nv0k32/nixconf/commit/728016afa01fe30952148d213a1d900f1cbdddcc). You'll need `CONFIG_CMD_SYSBOOT=y` for that command and the logging you added to `cmd/sysboot.c` to be compiled.

Once the flag's set, I'd try normal extlinux boot again. Your workaround also removes `usb start`, so I'd need a serial log to rule out a separate USB hang.

EDIT: Sorry @nvmd! If anyone else wants help, you can find me on the NixOS Matrix, or my email is in my profile.