nvmd/nixos-raspberrypi #87: RPi4: Wifi periodically dying

WizardUli, 2025-09-13:
Wifi on my Pi4 is dying every minute or two for a dozen of seconds with dmesg full of

```
[   11.880228] brcmfmac: brcmf_c_process_txcap_blob: no txcap_blob available (err=-2)
[   11.893472] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Aug 29 2023 01:47:08 version 7.45.265 (28bca26 CY) FWID 01-b677b91b
[   11.931522] Bluetooth: hci0: BCM: chip id 107
[   11.936581] Bluetooth: hci0: BCM: features 0x2f
[   11.942897] Bluetooth: hci0: BCM4345C0
[   11.946894] Bluetooth: hci0: BCM4345C0 (003.001.025) build 0000
[   11.960039] Bluetooth: hci0: BCM4345C0 'brcm/BCM4345C0.hcd' Patch
[   12.556890] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[   12.723161] Bluetooth: hci0: BCM: features 0x2f
[   12.729122] Bluetooth: hci0: BCM43455 37.4MHz Raspberry Pi 3+-0190
[   12.735466] Bluetooth: hci0: BCM4345C0 (003.001.025) build 0382
[   14.325547] 8021q: 802.1Q VLAN Support v1.8
[   15.307303] bcmgenet fd580000.ethernet: configuring instance for external RGMII (RX delay)
[   15.324569] bcmgenet fd580000.ethernet end0: Link is Down
[   22.060752] ------------[ cut here ]------------
[   22.065511] memcpy: detected field-spanning write (size 104) of single field "mf_params->data" at drivers/net/wireless/broadcom/brcm80211/brcmfmac/cfg80211.c:5643 (size 1)
[   22.081165] WARNING: CPU: 2 PID: 935 at drivers/net/wireless/broadcom/brcm80211/brcmfmac/cfg80211.c:5643 brcmf_cfg80211_mgmt_tx+0x62c/0x698 [brcmfmac]
[   22.095024] Modules linked in: 8021q garp brcmfmac_wcc btsdio cdc_acm hci_uart btqca brcmfmac btbcm bcm2835_codec(C) brcmutil bcm2835_v4l2(C) bcm2835_isp(C) bluetooth rpivid_hevc(C) v3d cfg80211 bcm2835_mmal_vchiq(C) v4l2_mem2mem videobuf2_vmalloc videobuf2_dma_contig videobuf2_memops ecdh_generic gpu_sched raspberrypi_hwmon rfkill videobuf2_v4l2 videodev crct10dif_ce snd_bcm2835(C) vc_sm_cma(C) ecc i2c_brcmstb videobuf2_common mc raspberrypi_gpiomem nvmem_rmem xt_conntrack nf_conntrack uio_pdrv_genirq uio nf_defrag_ipv6 nf_defrag_ipv4 ip6t_rpfilter ipt_rpfilter xt_pkttype xt_LOG nf_log_syslog xt_tcpudp nft_compat nf_tables sch_fq_codel tls atkbd libps2 serio vivaldi_fmap tun tap macvlan bridge stp llc fuse efi_pstore nfnetlink ip_tables x_tables vc4 snd_soc_hdmi_codec snd_soc_core snd_compress ac97_bus snd_pcm_dmaengine snd_pcm snd_timer snd xhci_pci xhci_pci_renesas drm_display_helper cec drm_dma_helper dm_mod dax nls_iso8859_1
[   22.179156] CPU: 2 PID: 935 Comm: wpa_supplicant Tainted: G         C         6.6.31 #1-NixOS
[   22.187828] Hardware name: Raspberry Pi 4 Model B Rev 1.1 (DT)
[   22.193755] pstate: 60000005 (nZCv daif -PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[   22.200835] pc : brcmf_cfg80211_mgmt_tx+0x62c/0x698 [brcmfmac]
[   22.206903] lr : brcmf_cfg80211_mgmt_tx+0x62c/0x698 [brcmfmac]
[   22.212963] sp : ffffffc0810cb660
[   22.216330] x29: ffffffc0810cb6c0 x28: ffffff8041759ba4 x27: 0000000000000068
[   22.223599] x26: ffffff804d4fd558 x25: ffffff804140ccc0 x24: 0000000000000080
[   22.230862] x23: ffffff804d4fc000 x22: ffffffdbe6376698 x21: ffffff804d4fc008
[   22.238105] x20: ffffffc0810cb760 x19: 0000000000000088 x18: 0000000000000000
[   22.245348] x17: 0000000000000000 x16: 0000000000000000 x15: 0000000000000000
[   22.252613] x14: 000000000000000e x13: 7720676e696e6e61 x12: 0000000000000000
[   22.259877] x11: 0000000000000000 x10: 0000000000000000 x9 : 0000000000000000
[   22.267140] x8 : 0000000000000000 x7 : 0000000000000000 x6 : 0000000000000000
[   22.274403] x5 : 0000000000000000 x4 : 0000000000000000 x3 : 0000000000000000
[   22.281664] x2 : 0000000000000000 x1 : 0000000000000000 x0 : 0000000000000000
[   22.288911] Call trace:
[   22.288915]  brcmf_cfg80211_mgmt_tx+0x62c/0x698 [brcmfmac]
[   22.296983]  cfg80211_mlme_mgmt_tx+0x1ec/0x510 [cfg80211]
[   22.302633]  nl80211_tx_mgmt+0x354/0x3e8 [cfg80211]
[   22.308396]  genl_family_rcv_msg_doit+0xdc/0x178
[   22.313685]  genl_rcv_msg+0x248/0x2d0
[   22.317398]  netlink_rcv_skb+0x68/0x158
[   22.321287]  genl_rcv+0x40/0x70
[   22.325175]  netlink_unicast+0x330/0x3a0
[   22.329847]  netlink_sendmsg+0x1dc/0x488
[   22.334345]  __sock_sendmsg+0x78/0xf8
[   22.338059]  ____sys_sendmsg+0x2b8/0x340
[   22.342035]  ___sys_sendmsg+0xb8/0x138
[   22.345836]  __sys_sendmsg+0x90/0x118
[   22.349551]  __arm64_sys_sendmsg+0x2c/0x58
[   22.353705]  invoke_syscall+0x50/0x160
[   22.357508]  el0_svc_common.constprop.0+0x48/0x130
[   22.362367]  do_el0_svc+0x24/0x50
[   22.365728]  el0_svc+0x48/0x190
[   22.368911]  el0t_64_sync_handler+0x140/0x150
[   22.373328]  el0t_64_sync+0x190/0x198
[   22.377039] ---[ end trace 0000000000000000 ]---
[   22.548874] ieee80211 phy0: brcmf_cfg80211_external_auth: auth_status iovar failed: ret=-52
[   36.111761] brcmfmac: brcmf_set_channel: set chanspec 0xd022 fail, reason -52
[   36.227688] brcmfmac: brcmf_set_channel: set chanspec 0xd026 fail, reason -52
[   36.343628] brcmfmac: brcmf_set_channel: set chanspec 0xd02a fail, reason -52
[   36.459676] brcmfmac: brcmf_set_channel: set chanspec 0xd02e fail, reason -52
```

Here is my last config I tried: https://codeberg.org/Uli/nixsauce/src/commit/27dd52e2e995b9671da5db7609c1aa6a266473d3/nixos-configurations/voron0/default.nix

I have tried both the default 6.12.25 and now even 6.6.31 kernel bundle and multiple Pi4 and they all behave same.
Am I missing some configuration?

nvmd, 2025-09-17:
Hi!
I'm not using WiFi on any of the Raspberry boards, so I can't really tell you anything from my experience, unfortunately. The configuration does look good at the first glance, though.

Here are some questions to understand the full picture better, maybe it'll help us to get to the bottom of the issue:
- Do the boards have the latest EEPROM firmware?
- Did you try the latest upstream kernel versions? stable 6.12.34 and unstable 6.12.44?
- Are those boards in metal cases or something like that, that could affect signal levels, maybe "breaking" the Wi-Fi more often that otherwise?

WizardUli, 2025-09-21:
> Do the boards have the latest EEPROM firmware?

Yes, stable release channel. (I think it was version from May 2025.)

> Did you try the latest upstream kernel versions? stable 6.12.34 and unstable 6.12.44?

I haven't yet but I will since I want to solve this.

>Are those boards in metal cases or something like that, that could affect signal levels, maybe "breaking" the Wi-Fi more often that otherwise?

No and they also all work with *Raspberry Pi OS* kernel 6.12.25. Weak signal also should no cause stacktrace dumps in dmesg.

I'm going to *slowly* investigate this. Can we keep this issue open in meantime? Perhaps it'll attract other people with Broadcom wireless problems.